dgiese/dustcloud #189: Trying to root the Roborock S50 3.3.9-001710

michel72, 2019-01-28:
Hi,

I'm trying to root the S50, but the firmware is nowhere to be found.
My unit has v3.3.9_001710 but the latest firmware I can find is https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_001702.pkg

Can anyone tell me where to get this version and if it can be rooted?

Thx!

cryptomilk, 2019-01-28:
Just flash v11_001702 ...

michel72, 2019-01-28:
I've done so now, but I'm having issues with the sound files:

```
compatible readlink found!
Generate SSH Host Keys
Generating public/private rsa key pair.
Your identification has been saved in ssh_host_rsa_key.
Your public key has been saved in ssh_host_rsa_key.pub.
The key fingerprint is:
SHA256:RarfKnLBkj2Y0+/v2tteNlgJ66d/30ylooZd384nklQ root@ubuntu-mini
The key's randomart image is:
+---[RSA 2048]----+
|          .      |
|         o       |
|        . . .    |
|       . .   oE. |
|     B. S   ..o .|
|    * *. . .oo ..|
|     o +.o.o+o*..|
|    . o ooooo*o==|
|     o o+=*o+o.+O|
+----[SHA256]-----+
Generating public/private dsa key pair.
Your identification has been saved in ssh_host_dsa_key.
Your public key has been saved in ssh_host_dsa_key.pub.
The key fingerprint is:
SHA256:Z5rC61KZ8QX1aPYWw1qcbZ9NhuPOhAjSN5F0B3sKkc0 root@ubuntu-mini
The key's randomart image is:
+---[DSA 1024]----+
|         o==o..  |
|       .. oOE= . |
|      . o.B X * o|
|      .. =.B O =o|
|       =S.= = o.o|
|     .+ .= . +   |
|     .o o     o  |
|    .  o         |
|     oo          |
+----[SHA256]-----+
Generating public/private ecdsa key pair.
Your identification has been saved in ssh_host_ecdsa_key.
Your public key has been saved in ssh_host_ecdsa_key.pub.
The key fingerprint is:
SHA256:8RQ4uio1rxG6sgIhKz+DIinLRP3uZk7MnmFCRnxOm10 root@ubuntu-mini
The key's randomart image is:
+---[ECDSA 256]---+
|         ..      |
|  .     o  .     |
|   o o ..E.      |
|o o + = .+       |
|.+ +.+ oS .      |
|= o.*..          |
|o=.o.@           |
|@ =.B++          |
|B=.+*B           |
+----[SHA256]-----+
Generating public/private ed25519 key pair.
Your identification has been saved in ssh_host_ed25519_key.
Your public key has been saved in ssh_host_ed25519_key.pub.
The key fingerprint is:
SHA256:THPZOSKxw2atsqW7xa1rfuupPkAXuEzEEgsBhZCajAw root@ubuntu-mini
The key's randomart image is:
+--[ED25519 256]--+
|+=+..+...        |
|E  ...+..+ o .   |
|*.  .+ .X.= +    |
|+o    +=.* . .   |
|     ...S        |
|      .* .       |
|      o.o .      |
|       oo...     |
|      o=**+.     |
+----[SHA256]-----+
decrypt soundfile
ccrypt: /home/michel/dc/english.pkg: key does not match -- unchanged
unpack soundfile

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
decrypt firmware
unpack firmware
patch ssh host keys
disable SSH firewall rule
integrate SSH authorized_keys
reconfiguring network traffic to xiaomi
cp: cannot stat '../sounds/*.wav': No such file or directory
pack new firmware
encrypt firmware
a4ea235710f7ba21f31cfa28e47f20f9  output/v11_001702.pkg

```

unpack soundfile

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

michel72, 2019-01-28:
I downloaded: https://github.com/dgiese/dustcloud/tree/master/devices/xiaomi.vacuum/original-soundpackages/encrypted

cryptomilk, 2019-01-28:
You downloaded it as raw? The log above clearly tells you that it can't read the file. It expects gzip but it is not a gzip file!

dgiese, 2019-01-28:
Btw: you do not need to integrate english.pkg anymore, as english is integrated in any new firmware. Just do the CCC2EU patch.

michel72, 2019-01-28:
Hi, I have the international version, but I was unable to execute the imagebuilder.sh without the english.pkg.

I now have successfully rooted. Indeed I made the stupid mistake to use the wrong github url for downloading the package.

Thanks both for your help!

closed 2019-01-28 by michel72