dgiese/dustcloud #193: Please make imagebuilder.sh reproducible

josch, 2019-02-07:
Hi,

currently, if two people run the same version of `imagebuilder.sh` with the same firmware file and the same options, they will receive a different modified firmware image. The differences are mostly (but not only) in the ssh keys that each of them generated independently.

Here my proposal: What if two people running the same version of `imagebuilder.sh` with the same original firmware input and the same options would get bit-by-bit identical output? The main benefit would be, that somebody who runs `imagebuilder.sh` could then use the hashsum (md5/sha/...) of the output they generated to verify, that the image they just built is indeed exactly the expected output of the script. This in turn will allow additional confidence of people to know that the image they just built will really work. A wiki page could list: run `imagebuilder.sh` with these options and that input and you will get a file with this checksum and if you flash your robot with a file with that checksum, then you probably didn't do anything wrong. It would also improve the ability to find and debug problems because currently, it's really tricky to say how somebody built a firmware image and it's close to impossible for somebody else to reproduce another person's steps. But if the output is bit-by-bit reproducible, then somebody could say: I used this input to get this firmware output with that hash and I'm running into these problems.

The biggest blocker right now, so that different people can produce bit-by-bit identical firmware images is, that ssh keys are generated on the computer running `imagebuilder.sh` and then directly integrated into the output image. This could be avoided, if instead of containing pre-generated ssh keys, the image would contain a startup script, which would generate the ssh keys upon first boot. But I don't know if this is feasible because maybe the used ARM platform does not provide enough entropy and then the boot process would take too long?

This is also known as [reproducible builds](https://reproducible-builds.org/docs/buy-in/) in Debian and Fedora and over there it is used mainly for security reasons. Here, it would more be for verifiability of the build process and debugging reasons but because of the existing reproducible builds efforts in Debian and Fedora, most tools today already come prepared to produce bit-by-bit identical output through the `SOURCE_DATE_EPOCH` environment variable. For example `tar` can be made to produce the same tar file that way.

Before I prepare a pull request that implements such a feature, I wanted to know your opinion and if you already know a blocker that makes this endeavour impossible.

cryptomilk, 2019-02-07:
I'm sorry, but the options will never be the same!

We are not an Open Source Linux distribution which wants reproducible build. We modify a firmware image. If you want a reproducible firmware image then use the original firmware and don't modify it.

Just for an example. I have a different OpenSSH public key then you do. If you want reproducible build you can't root your device and install an SSH key ...

josch, 2019-02-07:
I don't have a solution for the openssh public key problem. I didn't think about that. Closing the issue.

closed 2019-02-07 by josch