dgiese/dustcloud #194: Username / Password for Flolevac?

imbaimbaimba, 2019-02-16:
Hi everyone,

I just set this up, and it worked great. Also, I can work the robot from http://rockrobo now, great.

However, I have been having an extremely hard time with coordinates for zoned cleaning. I heard the Flolevac app was an easy way to do it. I downloaded it, logged in with my ip, token and did from /mnt/data/miio/device.conf
However, when I want to see a map in there, it asks me to set up server configuration. I just put the vacuums IP in the IP field, but I need a username and password. Haven't set any credentials for it AFAIK. Can anyone help me?

cryptomilk, 2019-02-16:
http://rockrobo/zone

imbaimbaimba, 2019-02-24:
> http://rockrobo/zone

Unfortunately the coordinate feature is not there anymore. So I still need a way to find coords :(

cryptomilk, 2019-02-25:
I think you're here in the wrong project. You should look into the Valetudo project.

However looks like Matthias did something in that direction now:
https://github.com/matthiasharrer/Valetudo/commits/zone-configuration-via-map

closed 2019-04-09 by dgiese