dgiese/dustcloud #199: Xiaomi Mi Gen1 automatic Downgrade

desq42, 2019-03-05:
Hi,

I presume this is not an issue with dustcloud, but perhaps someone has seen something similar.
I wanted to Root/Flash the newest Firmware to my Gen1 (3.3.9_003600.pkg)
Build and flash works like a charm. My Gen1 flashes and I can briefly access Valetudo 0.2.0 to
check the firmware. Everythings fine. 3-4 Minutes later, my Gen1 reboots and restores it´s old firmware. (3.3.9_3514) incl my previous Root & Valetudo. Has somebody a hint, how to prevent
this behavior? 

Thanks.
Desq

cryptomilk, 2019-03-05:
Did you follow: https://github.com/dgiese/dustcloud/wiki/Cloud-Free-Firmware-Image-With-Valetudo

desq42, 2019-03-05:
Yes, I did follow it.

cryptomilk, 2019-03-05:
The robot uses a A/B system. So it flashes to B and boots it, if everything is fine it flashes to A too, if not it boots to A.

This means something probably crashes and that's why it boot system A again. You probably have to flash again, connect via ssh as soon as possible and tail log files to find out what is happening.

desq42, 2019-03-05:
okay thanks. I´ll give it a shot.

desq42, 2019-03-06:
mh. Yesterday I flashed the original FW 3.3.9_003600, but after 3 Minutes it reverted itself back to my rootet & valetudo 3.3.9_003514. WTF?!  It seems as if I´ll have to do a factory-Reset, or has anybody an Idea? Which should I view to fix this issue?

cryptomilk, 2019-03-06:
FW 3.3.9_003600 is a beta version. The latest official version should be: 003468. Maybe you shouldn't use beta software :-)

desq42, 2019-03-06:
Good advice. ;-) Where do I get it? 
[image]

cryptomilk, 2019-03-06:
https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Firmware

desq42, 2019-03-06:
dammit. Same behaviour with v11_003468.fullos.pkg
Seems as if this guy is begging for a factory reset.

herrwusel, 2019-03-06:
I had the same behaviour and it was gone after a reset. Might be worth trying. Did you change anything on the robot like installing additional packages?

desq42, 2019-03-06:
except for midnight commander nothing else.
got to search this russian mi home app…

closed 2019-03-07 by desq42