dgiese/dustcloud #201: dustcloud.seemoo.de 403 forbidden

tmechen, 2019-03-12:
https://dustcloud.seemoo.de/public/xiaomi.vacuum.gen1/original/ is not reachable for firmware

MrWook, 2019-03-19:
So will this be fixed? Would love to use dustcloud...

d1slact0r, 2019-03-19:
Same issue here, please fix @dgiese

dgiese, 2019-03-19:
As as fallback, you can download the unpacked images from here: https://dustbuilder.xvm.mit.edu/templates/
I will check out whats going on. Thanks for the hint.

flindeberg, 2019-04-01:
@dgiese Any updates?

I'm looking for the 3468 firmware for gen1 but can't seem to find it anywhere (i.e. not on the mirror and not on the awsbj0 / awsde0 CDN hosts, get "Bucket not found" for bj0 and "Object not found" for de0).

dgiese, 2019-04-02:
Hi, sorry for that. Could not solve the problem with the old mirror (as I do not control the hardware).
I set up a new one here: https://dustbuilder.xvm.mit.edu/pkg/

flindeberg, 2019-04-02:
@dgiese Thank you a metric crap tonne! 

Do you happen to know if gen1 3468 is available somewhere? I can find 3466 on your new mirror.

closed 2019-04-28 by dgiese