dgiese/dustcloud #202: [FR] Add USB or Other Camera Support/Telepresense hack?

aenertia, 2019-03-12:
Has anyone successfully added a camera module to the gen2 vac? I see there is a USB storage mod in the Wiki, so presumably it would be relatively trivial to add USB(2?) camera to the unit. 

If there is interest in this it would be a great feature ; possibly even usable as a Video Conference Telepresense given a lot of the USB cam's have Mic's as well. Is it possible to route audio to the inbuilt speakers?

EliasKotlyar, 2019-03-16:
Hello @aenertia ,

I have tried, and i have failed. You can read my story here:
https://github.com/simonswine/rockrobo-kernel/issues/1

Routing Audio to Inbuild Speakers should be possible. You can play mp3s using:
sox file.mp3 -t wav - | /opt/rockrobo/firmware/uart_test -e /dev/stdin

Andoramb, 2019-04-10:
@EliasKotlyar I saw your attempts about the camera. Motion installs just fine, did you progress on uvc drivers? :)

EliasKotlyar, 2019-04-14:
@Andoramb Ok i will get a short Status-Update:
Compiling of UVC on the original 3.4.19 Kernel was not sucessful. There is a lacking of recent USB-drivers, which will support some high-speed modes needed for the webcam.

I have started compiling a mainline-u-boot and was sucessful with doing that. Now the U-Boot can start a newer kernel. Then i have encountered some issues with the MMC-Driver on the newer Kernel: It does start, however it cannot load rootfs. I was able to fix them by simple bruteforcing all of the possible combinations of the driver, and turning on "debug-mode" of the driver. Unfortunately, it only works in Debug-Mode - i suppose that it needs some different timings, which the debug-option adds(due to the serial output). I stopped progress for now, and i am waiting for some feedback on my issue in https://github.com/linux-sunxi/linux-sunxi/issues/294

Greatings 
Elias

Andoramb, 2019-04-15:
Thanks for the update :) I guess this is the situation when an issue brings 5 other ones :)

EliasKotlyar, 2019-05-19:
Hello,

I have discontinued the work on that project, because i cannot proceed. I have commited my patches here:
https://github.com/EliasKotlyar/RockRoboMainlinekernel

I hope someone can continue working on that...