dgiese/dustcloud #204: CE to CCC

akincansenol, 2019-03-22:
Hi all, 

I'm looking for a solution that needs to give me a possibility to connect my CE S50 and other products (Air purifier, bulbs etc) which they do not show up on Mi Home EU region. As far as I understand that I need to convert my S50 to CCC to gain this possibility but I couldn't find a way in the wiki, as there is just CCC to CE conversation tutorial.

Any help will be appreciated.

Best,
Akin

zvldz, 2019-04-12:
sed -i 's/location.*/location=prc/' /mnt/default/roborock.conf

zerenx, 2019-06-02:
On my International (CE) Roborock S50, using firmware 3.3.9.001768 and 3.3.9.001782 and then converting it to China (CCC) version makes the Mi Home app spout this error whenever I try to access the Roborock's menu :

你购买的机器入未经过认证无法使用插件,请从正规渠道购买，如有疑问请联系经销商

Which is saying the device in uncertified and tells me to contact my seller.

I wish to use this Roborock using the Mainland China server for the same reason as OP do.

pukkancs, 2019-10-18:
Hi, so what worked for me so far:

[Factory reset](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset) the device. (downgrade from whatever firmware you have as >1.5 rooting does not work)
[Update](https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion) with the latest rooted image. I just followed the linux instructions.
[Backup](https://github.com/dgiese/dustcloud/wiki/Files-to-be-backed-up)
Remove `roborock.conf`
```
mount /mnt/default/ -o remount,rw
rm /mnt/default/roborock.conf
```
Reboot (```reboot```)
Initiate WiFi setup and add it in the app logged in to China Mainland.

I haven't done firmware update yet (1.5.4) after rooting, will see if that breaks it or not.

dgiese, 2021-12-12:
For reference, check config files here: https://github.com/dgiese/robotinfo/tree/main/configfiles

closed 2021-12-12 by dgiese