dgiese/dustcloud #205: Rooting the gateway v3

prototux, 2019-03-23:
Hello,

Is it possible to root the gateway v3 like the vacuum robots? i've seen your talks and references to the NXP firmware, but it seems that the newer gateway actually run linux (on the i.mx6 SoC), so it should theorically be possible to gain root access to this one, but was it done already or is it some new unexplored road?

dgiese, 2019-04-02:
Yeah, technically I did it already and I am working on a public root method. Will take some time, as I had some setbacks...

prototux, 2019-04-02:
Can i help? (i do have a mieu01, not a v3, but it seems they're basically the same, but with some different configuration and/or /home/root/gw), i rooted mine using the serial console, but i will probably continue to work on a "upgrade root" like for the vacuum cleaners, as it's more future proof than some temporary modification that will be gone at the next update.

I'll also probably work on some reverse engineering to be able to use it offline, on a repo on https://github.com/wiecosystem probably, it will change from BLE devices a bit :)

dgiese, 2019-04-02:
send me a mail or a pm via telegram

prototux, 2019-04-02:
I've sent you an email (i don't really use messaging apps)