dgiese/dustcloud #208: imagebuilder how to activate soundfiles?

fvollmer, 2019-04-08:
I build an image (with the newest image builder available):
`sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh --public-key=/home/fvollmer/.ssh/id_rsa.pub --soundfile=../firmware/english.pkg --firmware=../firmware/v11_003468.fullos.pkg --valetudo-path=../valetudo --disable-firmware-updates --replace-adbd
`
but the robot still speaks chinese. The soundfiles were actually copied to `/opt/rockrobo/resources/sounds/en`. As a quick hack I just moved the folder
```
mv prc/ prc_/
mv en/ prc/
```
and now the robot speaks English. Is there a better way?

CrazYoshi, 2019-04-08:
> I built an image (with the newest image builder available):
> `sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh --public-key=/home/fvollmer/.ssh/id_rsa.pub --soundfile=../firmware/english.pkg --firmware=../firmware/v11_003468.fullos.pkg --valetudo-path=../valetudo --disable-firmware-updates --replace-adbd `
> but the robot still speaks chinese. The soundfiles were actually copied to `/opt/rockrobo/resources/sounds/en`. As a quick hack I just moved the folder
> 
> ```
> mv prc/ prc_/
> mv en/ prc/
> ```
> and now the robot speaks English. Is there a better way?

I didn't get where get these folders

cryptomilk, 2019-04-09:
Normally the robot language should be set to English, then the english sound pack which is installed on the robot is used.

Valetudo -> Settings -> Info -> Language: English

It might be Chinese if you configured a chinese server instead of a European one. However just install a sound pack and the problem will be fixed as those are installed in /mnt/data

cryptomilk, 2019-04-09:
`mirobo raw-command app_get_locale` will also return the language

fvollmer, 2019-04-09:
Location and language are set to PRC. I assume this is the default. I never
connected the robot to the Xiaomi Cloud.
How can I change the language?

cryptomilk, 2019-04-09:
My location is set to de and so the language is en. You could try:

`echo "country_code=DE" > /mnt/data/rockrobo/country_info`

on the robot as root.

fvollmer, 2019-04-09:
I just tried this and also did a reboot, but it didn't change the location or language displayed in valetudo.

cryptomilk, 2019-04-09:
Try configuring the german log server maybe that will change it ...

fvollmer, 2019-04-09:
I'm not sure how to change the log server. Where can I change it and what
would be the new addrress? I assume `awsbj0.fds.api.xiaomi.com` is the PRC
log server?

cryptomilk, 2019-04-09:
I think you have to edit `/mnt/default/roborock.conf` there are all the settings.

Franzl123, 2019-04-09:
see https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion

but how can we do it while firmware building?
on the robot, "/dev/mmcblk0p6" is mounted to "/mnt/default/"

maybe something like an "one time script" on first boot via "/etc/rc.local"?

fvollmer, 2019-04-09:
Thanks! I just successfully changed the language by creating the `/mnt/default/roborock.conf` file for a Gen1 robot. Not only `en` works, but I also could use `de`! (The robot plays the soundfiles from `/opt/rockrobo/resources/sounds/de`)

closed 2019-04-09 by fvollmer