dgiese/dustcloud #211: Gen1 vacuum, 3.3.9_\d{6} vs 3.5.0_\d{6} line of firmware

flindeberg, 2019-04-12:
Roughly a week ago 3.5.0_003476 was released for Gen1, although I haven't upgraded due to lack of time to flash to older version if something breaks. I'm worried that Xiaomi / Roborock is going to shut down modability sometime soon. 

Has anyone tried to upgrade to the 3.5.0 line of firmwares and can report success with dustcloud, mirobo, rooting, etc?

For Gen1 I can explicitly upgrade from 3.3.9_003468 to 3.5.0_003476.

kikettas, 2019-04-16:
I have the latest update (`3.5.0_003476`) and I couldn't send the firmware to the vacuum. I specifically get this error: 
```
unable to parse json '': Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "dustcloud/devices/xiaomi.vacuum/firmwarebuilder/flasher.py", line 207, in <module>
    main()
  File "dustcloud/devices/xiaomi.vacuum/firmwarebuilder/flasher.py", line 185, in main
    r = vacuum.send('miIO.ota', ota_params)
  File "/Users/enriquedelpozogomez/Desktop/rockrobo/flasher/venv/lib/python3.7/site-packages/miio/device.py", line 265, in send
    self.__id = m.data.value["id"]
TypeError: 'NoneType' object is not subscriptable
```

flindeberg, 2019-04-16:
Just speculation here, but then it seems like Xiaomi really did something when going from 3.3.9 to 3.5.0. Perhaps connected to the T6 release? 

Does mirobo work?

kikettas, 2019-04-16:
It didn't. Just reset the vacuum to factory settings (Pressing `Home` + `Reset` button), and from the stock firmware I could upload the firmware to the device.

So yes, there might be an issue with `3.5.0`

fvollmer, 2019-04-16:
The `3.5.0_003476` update package could be very interesting. This update package should be encrypted with the old key and might use the new update mechanism of the T6. This could help #213.

Nixon506E, 2019-04-18:
The T6 I have is currently running `3.4.5_0412` and says that is the most up to date version. We have obviously seen that version number doesn't necessarily mean newer for these devices but it is not necessarily the same branch or encryption as `3.5.0_003476`.

cherokee93, 2019-05-11:
Yesterday I did a big mistake, I upgraded my vacuum to 3.5.0 through the app. Now ssh is not working anymore, Valetudo is installed (but [showing](https://i.imgur.com/CRj489z.png) old  3.3.9 version), I can use Valetudo and the Xiaomi App, but I get the  error above when trying to upload custom firmware.

For now I can't do anything, right?

kikettas, 2019-05-13:
For me resetting the firmware to factory settings allowed me to install Valetudo, I had the same issue you are having with `3.5.0`

cherokee93, 2019-05-13:
Uhm, how did you reset to factory settings? The instructions on the wiki are not working for me.

kikettas, 2019-05-13:
Hold Home Button and press WiFi-reset button, continue holding the home button until reset notification is played.

br34k0r, 2019-05-13:
I also have the same problem with a fresh 3.5.0 updated robot (1st gen). So doing a factory reset allows to flash valetudo?

cherokee93, 2019-05-13:
> Hold Home Button and press WiFi-reset button, continue holding the home button until reset notification is played.

How many seconds did you hold? For me it never worked and I have tried few times. 🤔

br34k0r, 2019-05-13:
@cherokee93 I think first gen factory reset is  RESET+Home

cherokee93, 2019-05-13:
> I think first gen factory reset is RESET+Home

Thanks a lot! I hope it works!
If that's the way, then the documentation is wrong.
When I return back home I'll try 🤞

dugite-code, 2019-05-13:
@cherokee93 I updated the wiki after I had issues resetting my 1st gen [Xiaomi-Vacuum-Robots-Factory-Reset](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset)

> Hold Home Button and press WiFi-reset button, continue holding the home button until reset notification is played.

kikettas, 2019-05-13:
You're right @br34k0r, it's not how I posted before but how you describe 👍

cherokee93, 2019-05-13:
I don't know, for me home+wifi-reset+keep-holding-home-while-releasing-wifi didn't work. 🤔

dugite-code, 2019-05-13:
Yeah it took me a quite while to get it just right. It's surprisingly fiddly to get it to work. It's about 5-10 seconds to get the notification if memory serves

cherokee93, 2019-05-13:
> Yeah it took me a quite while to get it just right. It's surprisingly fiddly to get it to work. It's about 5-10 seconds to get the notification if memory serves

Thanks, I'll do more tests and report back. So, after resetting it restores the image from the recovery partition?

dugite-code, 2019-05-13:
Yes, it's a complete wipe mine for example goes back to speaking Chinese and I have no ssh access.

cherokee93, 2019-05-13:
> Yes, it's a complete wipe mine for example goes back to speaking Chinese and I have no ssh access.

Perfect 👌 
It feels super weird to have Valetudo and no ssh access, I want to start from clean.

By the way, will they ever fix the map being rotated randomly every time it genereates the map? All my automations in homeassistant have no point 😞

dugite-code, 2019-05-13:
Basically the best way to avoid map rotations is you will be waiting on save map feature to be back-ported from the Gen2. As it's still getting updates it's possible it will happen.

Until then you can do what I used to do and avoid doing a general clean. Only use zone clean-up's as that doesn't alter the active map. I had the entire house zoned in homeassistant complete with a few goto commands to stop it from getting lost.

br34k0r, 2019-05-13:
@cherokee93 Can you give me feedback, if it works with flashing? I tried the whole weekend with mine and didn't get it working.

Thank you

cherokee93, 2019-05-13:
> @cherokee93 Can you give me feedback, if it works with flashing? I tried the whole weekend with mine and didn't get it working.
> 
> Thank you

Sure! I'll report back in max 1-2 hours. 

> Basically the best way to avoid map rotations is you will be waiting on save map feature to be back-ported from the Gen2. As it's still getting updates it's possible it will happen.

I really hope so, I love Xiaomi, because they keep supporting older devices. I hope this features will eventually come, also the virtual wall thingy (my vacuum loves to eat my curtains)

> Until then you can do what I used to do and avoid doing a general clean. Only use zone clean-up's as that doesn't alter the active map. I had the entire house zoned in homeassistant complete with a few goto commands to stop it from getting lost.

Something I don't understand, is when and why the map gets regenerated. 
If I pick up the robot it gets regenerated. If I do full cleanup it stays the same 🤷‍♂

cherokee93, 2019-05-13:
@br34k0r It worked out! I pressed home+power, after the wifi reset message, I've press the reset pin hole. After some seconds, I release the reset pin, keep holding the Home button and It said something like "resetting to initial configuration, it can take 5 mins" . BINGO

Then I connected my laptop to it, retrieved the token, tried to flash the firmware, it gave me error because it was undocker. Docked it, retried and it gave me the same error _ERROR:miio.updater:No request was made.._ . I retried to input the same command and it worked 🤷‍♂ 

BONUS: I can't ssh to it  because I f'd up something with firmware generation and now I don't remember the password 🤔 So far so good, Valetudo works, Xiaomi app worked with setting up wifi (map is not showing because of Valetudo hosts).

Thanks a lot @dugite-code @kikettas 🙏 👏

dgiese, 2019-05-13:
@cherokee93 The root password is being changed while every update and factory reset. If you backed up the device configuration (device.conf and vinda), you can login with the content of the vinda file (XOR '6') as the root password. Otherwise you should have a ssh key...

cherokee93, 2019-05-13:
@dgiese yeah I see. 
I'm trying to understand how ssh keys work (I'm really noob). 
Some days ago, after I succesfully logged in, I changed password with passwd, then I could login from any of my computers.

> If you backed up the device configuration (device.conf and vinda)
Next thing I'm going to do when I'll be able to access ssh. I didn't do it last time and now I "kinda" understand how it works thanks to[ this issue](https://github.com/dgiese/dustcloud/issues/81)

dgiese, 2019-05-13:
@cherokee93 Changing the root password with passwd is only a temporary thing. As soon as you update or reset the vacuum, it gets reset to the obfuscated vinda password.

Maybe your build process is broken and you add the wrong ssh public key. You could try to get a working image using https://dustbuilder.xvm.mit.edu and let it generate the keys for you.

cherokee93, 2019-05-13:
> Maybe your build process is broken and you add the wrong ssh public key. You could try to get a working image using https://dustbuilder.xvm.mit.edu and let it generate the keys for you.

Yep I think that my problem was the wrong public key path.
Now it works 👍 

I'm backing up device.conf and vinda. 
So, hardware-generated root password is in the vinda file ? 
Guess it's time to follow the **"Important files to back up"** steps from the wiki. 

**Thanks a lot for this project** 🥇

gerstemann, 2019-05-17:
How do I get the update or image? Home app is saying vaccum is up to date using the 3600.

gerstemann, 2019-05-29:
Hi there, found a solution and could flash the new firmware. If anyone interested, can provide files to flash. Worked for my vaccum without any error. Mi Home app is showing the new version. no changes so far.

akeilox, 2019-05-30:
how did you do it?

flindeberg, 2019-05-30:
@gerstemann, which firmware? And root as well?

gerstemann, 2019-05-31:
I mean the new 3.5.0_003476 für Gen1 vacuum.
I used the differential firmware image available here: https://dustbuilder.xvm.mit.edu/pkg/v1/3.5.0_003476/ and mirobo under windows os.

mahirgul, 2020-02-15:
3.5.0.xxxx root password is "roborock".

flindeberg, 2020-02-26:
> 3.5.0.xxxx root password is "roborock".
> Mi robot gen1 only

Have you managed to get duscloud / similar to work?

mahirgul, 2020-02-26:
I cant because my device is not working properly. 
My device has not been working as expected since I installed this version. and I was never able to return it to the factory version.
But I'm still working on it. I even attached an esp8266 on it to provide a terminal connection while the device was running around.

EliasKotlyar, 2020-04-13:
I can confirm that the rooting on the 3.5.4 Firmware on Gen 1 does not work. Workaround is to return back to stock firmware using factory reset and then root the device as usual.

lupin-de-mid, 2020-04-19:
My Gen1 vaacum  has 4006 as stock firmware :(

flindeberg, 2020-04-19:
> My Gen1 vaacum has 4006 as stock firmware :(

Is the the refurbished Gen1? Can you downgrade firmware?

lupin-de-mid, 2020-04-19:
Probably it's new version. Not sure.  if after upgrade to 4007 reset through home+reset it return 4006

Signature%20placeholder

enykeev, 2020-05-06:
> My Gen1 vaacum has 4006 as stock firmware :(

Same problem here

r4nd0mbr1ck, 2020-05-15:
Sorry for another newbie question. I was previously on the 3.5 firmware but managed to factory reset and root the v11_003468.fullos.pkg (direct CDN) version.

Before I try and mess this up: Can someone confirm if root can be applied to the 3.5 firmware (from https://dustbuilder.xvm.mit.edu/pkg/v1/) using firmwarebuilder? 

Note: I am currently on rooted 3.3 so uploading the firmware won't be an issue. It's a question of whether the downloaded 3.5 can be rebuilt with root. 

The cleaning algorithm in 3.5 seemed a lot nicer so would like to have it.

flindeberg, 2020-05-16:
@r4nd0mbr1ck it seems like no one had gotten it to work. If you do get it to work do tell us.

I have not tried the 3.5 line of firmware at all.

dominik-ziegler, 2020-05-20:
> > My Gen1 vaacum has 4006 as stock firmware :(
> 
> Same problem here

Yeah same problem here. Even after factory reset reset, I'm stuck with the 4006 firmware.

Never contacted to Wifi-before/App before. 
Brand new Gen 1. 

Any ideas?