dgiese/dustcloud #213: Gen 3 (T6) Root

Nixon506E, 2019-04-16:
Just got a T6 and was hoping to help get the T6 verified as rootable. I believe the first step would be getting a copy of the latest firmware but am not sure where to look without connecting the vacuum to the internet and reading its request urls. Any suggestions?

dgiese, 2019-04-16:
Rockrobo changed the encryption of the Firmware+Soundfiles and some of the update process. As I don't have a T6, I cannot really help. So its at the moment not rootable.

Nixon506E, 2019-04-16:
Any suggestions on what would be a good next step considering I do have one. Hoping to help the community if I can.

fvollmer, 2019-04-16:
You may find https://dgiese.scripts.mit.edu/talks/34c3-2017/34c3.html interesting. IIRC the initial rooting was done by shorting pins with aluminium foil. So a pretty invasive method.

Nixon506E, 2019-04-16:
I have seen @dgiese's talk on this but it seems like the encryption key was kind of just stumbled upon? Not really looking to dump the entire flash of my device especially if it might not have the same passwords in plaintext as before.

fvollmer, 2019-04-16:
I'm sure @dgiese can answer this better, but I think the encryption key was found by analyzing the flash dump (which wasn't encrypted).

Nixon506E, 2019-04-16:
@fvollmer yes it was, but it was stored on the operating system in plain text and was being used as a password for something else, seems like it just happened to be the same for the ota encryption as well

dgiese, 2019-04-16:
Yeah, back then I extracted it by reverse engineering the binaries (its not stored in plaintext). However I think they learned from their mistakes and are doing things now differently. My usual approach would be to open the device and de-solder the MMC, and dump it. That has of cause the risk that you have a bricked device...at least if you are not familiar with BGA chips. But that's purely speculation, as I don't have the device. Time will show. I mean it took 1,5 years for Gen1 until someone, me, published the stuff. Guess for T6 it will be a little bit faster, but still someone needs to sacrifice his/her device.

Nixon506E, 2019-04-16:
Gotcha, not sure I am the right person to reverse engineer a password out of the new binary. I got a T6 pretty easily so they are obviously becoming more available. @dgiese, so you don't suggest we try to grab the flash through the usb port and FEL mode anymore? Hoping to not need to modify any hardware if I can get around it.

dgiese, 2019-04-16:
Honestly, no idea. If you disassembled it, make some pictures of the hardware. Would like to take a look at it.

zvldz, 2019-04-16:
Does anyone have a link to the encrypted firmware?

dgiese, 2019-04-16:
Maybe I know someone. But that's a discussion, which we should not have publicly *yet*. Thats more of a thing I would discuss in telegram or slack or something.

MickL, 2019-06-15:
The T6 is now available in the EU known as the S6 in Germany, France and Spain. Would really really appreciate rooting this device! 

Any news on this?

dgiese, 2019-06-26:
Unfortuantly they started to sign all firmware files and soundpackages. In addition they changed a lot of other things. One thing which is especially nasty is the fact, that now the region is set in a signed config file, which is bound to the cpu id of the device. So there is no trivial way to change the region (t6->s6). Also there is no simple way to root the device like v1 or s5. Beware: some of the nasty stuff seems to be backported to s5 models produced after march 2019. See my twitter comment for that.

MickL, 2019-06-26:
Thanks for the update! Sounds bad so far, but if the Nintendo Switch is hackable i have a lot of hopes! I ordered the S5 and hope it is not affected of the march update as you said.

Please tell us how we can support you. Donations, Patreon, etc. I want to help where i can.

dgiese, 2019-06-26:
A root of the S6/T6 and even for M1S is possible. However it is not as easy as the existing method for the Mi Vacuum Robot (aka Gen1 aka v1) or Roborock S5x (aka Gen2). Especially as it requires opening the device at the moment. I explore with some other peoples at the moment a different approach, however, if you want to have an easy rootable device, you should stay away from S6/T6 and M1S in particular (as M1S uses a completely different platform). The only difference between S5 and S6 seems to be a different brush and an additional sensor for the water tank. More information can be found here: https://www.roboter-forum.com/index.php?thread/35506-hack-des-roborock-s6-t6-planung-ger%C3%A4tekauf-fortschritte/&postID=474142#post474142

For S5: If you buy an global model, you should have no problems with the geoblocking. Also, if you use only valetudo, you should be OK even with the chinese model. The geoblocking is only in effect in combination with the Mi Home app + Cloud.

I am always happy about support for the procurement of new devices (as potential alternatives to v1 and s5). I did a fundraising for the T6 which went okayish. At the moment I look also at other vendors. If you have too much $$$, then drop me an email (dgi[at]mit.edu) and we find a way ;)

kiekerjan, 2019-08-14:
> Beware: some of the nasty stuff seems to be backported to s5 models produced after march 2019.

Which nasty stuff do you mean? Does the newer version not accept the rooted firmware?

dgiese, 2019-08-15:
They backported the Geoblocking mechanism of the T6 to newer S5. I cant really tell how it is exactly implemented on the newer S5, as I dont have one.

apll64, 2019-09-15:
@dgiese Any chance you could provide some insight to the method you are referring to in your comment https://github.com/dgiese/dustcloud/issues/213#issuecomment-505842643? I am currently thinking about buying a S6/T6 and would not mind opening the device. However, I need to be sure it is indeed rootable. I would prefer the S6/T6 because it supposedly charges quicker and makes a little less noise.

breichldomico, 2019-11-09:
> @dgiese Any chance you could provide some insight to the method you are referring to in your comment [#213 (comment)](https://github.com/dgiese/dustcloud/issues/213#issuecomment-505842643)? I am currently thinking about buying a S6/T6 and would not mind opening the device. However, I need to be sure it is indeed rootable. I would prefer the S6/T6 because it supposedly charges quicker and makes a little less noise.

I already own an S6 and would not mind opening the device if there is a way to get Valetudo on the device. If you could provide some hints it would be awesome! ;)

dgiese, 2020-07-02:
It is now supported, however you need to disassemble the devices :( I did not find a trivial way to root the device without disassembling. All options require some amount of it...
See my message here with the links to the tutorials.
https://twitter.com/dgi_DE/status/1277223172862029824

dgiese, 2020-07-02:
As a sidenote: The current image builder is not compatible with the S6/T6 due to the different firmware package format.

monster400, 2020-07-02:
Hello do you think this will work with the S5 max also? i added pictures in #252

jwveldhuis, 2020-07-02:
@dgiese would this work with the Roborock S4 too?

dgiese, 2020-07-02:
S5 Max is different. S4 might work... do you have one? then contact me ;)
Btw: I created an overview of the hardware to give you an idea: https://dontvacuum.me/robotinfo/

jwveldhuis, 2020-07-02:
@dgiese yes I have an S4, will send you an email.

daihashi, 2020-07-02:
> S5 Max is different. S4 might work... do you have one? then contact me ;)
> Btw: I created an overview of the hardware to give you an idea: https://dontvacuum.me/robotinfo/

The S4 is the model I disassembled and sent you pictures at the beginning of January. Not sure if you were ever able to look at the pictures I sent you. I sent a follow up email on March 18th, but I suspect you are very busy with other projects and may not have seen it.

dgiese, 2020-07-02:
Oh, sorry about that. I will check it. March was a very busy month due to a competition and c'na. If you have some time, we could run a quick beta test for the root and valetudo. Can you ping me again?

xsasx, 2020-07-14:
Does it mean that a 1S is now also rootable ?

cbergmann, 2020-07-22:
@dgiese I just got my S5 Max. Is there a Rooting method available? I don't have a big issue with opening the device. BGA soldering is another story.

dgiese, 2020-07-23:
Soon (TM). Ping me if you *really* need to root the device. At the moment we are still working on the tooling and fix the bugs.

cbergmann, 2020-07-23:
Hi, Thanks for the update and the work. I got it connected to my home assistant and that was the first goal. Disconnecting it from the cloud would be a nice addon but is not time critical. If I can help somehow (e.g. beta-testing tools or procedures) don't hesitate to contact me. You can also reach me via mail (dustcloud[at]cbergmann.net) in english or german.

bb12489, 2020-07-25:
Watching this thread with great enthusiasm! I'm hoping to pickup an S5 Max soon and would love to root it.

jgladch, 2020-08-04:
Subscribing for updates - looking forward to rooting my S5 Max 👍  Thank you for the work

dgiese, 2020-08-04:
The S5 Max video is delayed as my device is bricked :( This happened while I was testing the tools and methods (to that you guys don't brick your devices). There is a chance that the NAND flash is faulty... and I obviously cannot really send it in for warranty ;) It will take me some time to repair it.

JaanTaponen, 2020-08-09:
@dgiese Could we donate somewhere/somehow to help you get a new one?🤔

dgiese, 2020-08-11:
I hope that I don't need to buy a new one. I hope that at some point I have some free financial resources to just buy the PCB. If you really have too much $$$, you can ping me via mail ;) But don't worry. Eventually I find a solution.

Miph120, 2020-09-11:
@dgiese Just got an S5 Max as well. Let me know if there's anything I can do to assist. Thanks!

Trilis29, 2020-09-12:
Waiting also 👍

pit34, 2020-09-12:
Got an S5 Max too. Waiting and can help too.

DrDragonKiller, 2020-09-13:
Will the root require a certain firmware?  Or are we free to upgrade to the latest one? (Asking as there are Nintendo Switch Roots that don't work anymore on newer firmwares) :wink:

dgiese, 2020-09-13:
So I ordered a new board and it technically arrived a week ago. However, the package got lost somewhere in the university mail center (there is a lot of packages due to the new undergrads moving in). So I need to wait until they find it :(

dgiese, 2020-09-13:
> 
> 
> Will the root require a certain firmware? Or are we free to upgrade to the latest one? (Asking as there are Nintendo Switch Roots that don't work anymore on newer firmwares) 😉

As far as I can tell, all firmwares for T6 are rootable. However, newer versions are little bit more tricky. So far one can always factory reset to a older version.

stecky, 2020-09-23:
ooohh ooohh me too! S5 Max root please! :)

dgiese, 2020-09-23:
Contact me over Telegram if you dare to test drive the S5 Max root. I fixed my S5 Max again, and created a PhoenixSuit image in case that happens again to me ;)

stecky, 2020-09-24:
@dgiese has anyone taken you up on your offer? I don't really want to be the first... lol

cbergmann, 2020-09-25:
I would be interested depending on how risky/technical the process is.

ektooo, 2020-09-26:
Is there any chance/possibility that in the distant future it will be possible to root the S5 Max without having to tamper with the hardware? 

Has the installation of the sound packages been fundamentally changed or is only the encryption different? If the latter is the case: Do you already know which encryption and which key was used?

btw: Thank you for your commitment, your effort and your ambition! 👍💘  @dgiese

dgiese, 2020-09-26:
Hi, I tried the root with some peoples and we run again into issues of softbricking. We are debugging it at the moment. So please stay tuned.
I dont think that there will be a tamperless solution, as the firmware is signed and the update mechanisms are disabled.

alexwaibel, 2020-10-02:
Commenting to subscribe. Thank you so much for your work! Seems you have plenty of s5 max owning volunteers already but feel free to reach out if you need more.

daduke, 2020-10-07:
Commenting to subscribe

boesing, 2020-10-07:
> Commenting to subscribe

there is a subscribe Button ffs!!!

sibero80, 2020-10-28:
This thread is very exciting, commenting to stay subscribed. I own a S5 Max and know my basics in programming and electronics. If there is any way I can help, provide some documentation I can follow along and I'll do my best.

FireFrei, 2020-10-28:
Additional information for the rooting issue of my new S6:

After factory reset it seems to keep the newest firmware:
3.5.8_2076.

Is there any other way to change or remove the root password?

```
sunxi#ext4load mmc 2:7 40008000 /etc/OS_VERSION
Loading file "/etc/OS_VERSION" from mmc device 2:7
232 bytes read
sunxi#md 40008000                              
40008000: 702e6f72 75646f72 642e7463 63697665    ro.product.devic
40008010: 41543d65 5f534f4e 37303256 30322d36    e=TANOS_V2076-20
40008020: 36303032 31303332 5f4c4552 352e334d    20062301REL_M3.5
40008030: 545f382e 2e312e34 5f312d34 454c4552    .8_T4.1.4-1_RELE
40008040: 5f455341 30323032 33323630 3536312d    ASE_20200623-165
40008050: 0a313338 622e6f72 646c6975 7369642e    831.ro.build.dis
40008060: 79616c70 3d64692e 4f4e4154 504d5f53    play.id=TANOS_MP
40008070: 3631525f 4c45525f 45534145 3230325f    _R16_RELEASE_202
40008080: 32363030 36312d33 31333835 2e6f720a    00623-165831.ro.
40008090: 2e737973 74757063 3d657079 2e363152    sys.cputype=R16.
400080a0: 334d5453 33412e32 0a31472e 622e6f72    STM32.A3.G1.ro.b
400080b0: 646c6975 7265762e 6e6f6973 6c65722e    uild.version.rel
400080c0: 65736165 3032563d 720a3637 75622e6f    ease=V2076.ro.bu
400080d0: 2e646c69 65746164 6374752e 3935313d    ild.date.utc=159
400080e0: 32303932 0a313137 2e646170 2f74656e    2902711.pad.net/
400080f0: 6e756275 222f7574 424f520a 434f524f    ubuntu/".ROBOROC

sunxi#ext4ls mmc 2:7 /opt/rockrobo/firmware/img
<DIR>       1024 .
<DIR>       1024 ..
          217072 RR_TANOS_MP_STM32.A3.G1_M4.0.0_RELEASE_20200623-164315_FULL.bin
```

sibero80, 2020-10-29:
https://samsclass.info/123/proj10/p12-hashcat.htm

If you can post the password hashes, most likely we can get the PWD. I have a friend that will soon receive a Nvidia 3080, that could help speed up the process.

FireFrei, 2020-10-30:
Finally, after two long nights and half a day I got the robo with initial version 3.5.8_2076 rooted! It seems like the password is kind of auto generated and login is prohibited in general. At least I couldn't figure out, what the exact issue was, so why logging in (even with replaced and/or removed credentials) was not possible. But I found a way how to root it and were finally able to flash it.
I don't know if it is a good idea to post the new way here publicly, maybe not every steps are necessary (or will be "fixed-out" in the next official release of roborock). @dgiese if you are interested, I can provide my notes for future patches.

Has anyone experiences with flashing the recovery partition? My solution would not survive a factory reset and I want to avoid opening it again. Unfortunately I experienced some ssh issues before flashing the Dustcloud/valetudo image, so a simple sed in the recovery partition to permanently open the ssh ports may not help.

dgiese, 2020-10-30:
Hi,
just ping me via mail/Telegram.
Rockrobo introduced some time ago their custom login shell "rr_login" which does use its own authentication file. They also patched the SSH server, that why we are replacing it anyway. Its a little bit annoying that they do that. I have an alternative method, so we should be fine for now. But it gets more and more annoying.

sibero80, 2020-10-30:
That's great news! I hope the method also works on a S5 Max. Staying at the current firmware for the time being.

dgiese, 2020-10-30:
The S5 Max is another sad story, due to its different architectue.

andreas-bulling, 2020-11-14:
sad story in the sense of "won't be possible"? I hope not...

ekeih, 2020-11-15:
@FireFrei @dgiese Do I understand your discussion correctly, that the method described in https://www.youtube.com/watch?v=sn1tia80nT4 does not work anymore with new S6 models?

[61 more replies]

closed 2021-12-12 by dgiese