dgiese/dustcloud #215: imagebuilder.sh: loop is a module

ivan98, 2019-04-23:
My CONFIG_BLK_DEV_LOOP is compiled as a module, so I can't mount -o loop without first modprobing the module.

Suggest "modprobe --quiet loop" be inserted before the mount, as follows:

```
$ diff -Naur imagebuilder.sh.orig imagebuilder.sh
--- imagebuilder.sh.orig        2019-04-23 10:37:26.809665307 +0800
+++ imagebuilder.sh     2019-04-23 10:37:42.660092263 +0800
@@ -340,6 +340,7 @@
     #ext4fuse disk.img image -o force
     fuse-ext2 "$FW_DIR/disk.img" "$IMG_DIR" -o rw+
 else
+    modprobe --quiet loop
     mount -o loop "$FW_DIR/disk.img" "$IMG_DIR"
 fi
 ```

dgiese, 2019-04-28:
is this a potential thing which might break something for someone?

ivan98, 2019-04-28:
Not that I know of. If loop is built into the kernel, modprobe loop will yield loop module not found, which is harmless (though you may want to pipe the warning message to null, to not frighten people). For kernels with loop module but loop module is already loaded, it does nothing.

So either way, it is harmless.