dgiese/dustcloud #217: flasher.py not working on firmware 3.3.9_001748

Philipp97714, 2019-04-26:
I tried to root my Roborock S50 CN version with firmware 3.3.9_001748.
I downloaded v11_001820.fullos.pkg and de.pkg and patched the firmware. I disabled my firewall. My console output looks like this:

`python3 flasher.py -a 192.168.178.10 -t 5556525432473565546b4c396765576a -f output/v11_001820.fullos.pkg
Flasher for Xiaomi Vacuum
Connecting to device 192.168.178.10...
Starting local http server...
Serving http server at 172.17.74.99:42177
Sending ota command with parameters: {"mode": "normal", "install": "1", "app_url": "http://172.17.74.99:42177/v11_001820.fullos.pkg", "file_md5": "b12ffc8265caf4934a31208222e173b5", "proc": "dnld install"}
Ota started!
                                                                      
Firmware downloaded successfully.
Exiting.
`
The power button starts blinking but turns off again after showing "Firmware downloaded successfully." The processbar stucks at 0% and never increases so I guess nothing gets copied onto the roboter.

fvollmer, 2019-04-26:
This sounds strange. You could try to use `mirobo --ip yourip --token yourtoken firmware.pkg`.

Philipp97714, 2019-04-26:
> mirobo --ip yourip --token yourtoken firmware.pkg

I added "update-firmware" to your command. I get this error:

```
root@philipp-Virtual-Machine:~/dustcloud/devices/xiaomi.vacuum/firmwarebuilder# mirobo --ip 192.168.178.10 --token 5556525432473565546b4c396765576a update-firmware output/v11_001820.fullos.pkg
Going to update from output/v11_001820.fullos.pkg
INFO:miio.updater:Serving on 0.0.0.0:41255, timeout 10
INFO:miio.updater:Using local output/v11_001820.fullos.pkg (md5: b12ffc8265caf4934a31208222e173b5)
Hosting file at http://172.17.74.99:41255/v11_001820.fullos.pkg
Update started!
Downloading:   0%|                                      | 0/100 [00:09<?, ?it/s]ERROR:miio.updater:No request was made..
Failed: 100%|█████████████████████████████████| 100/100 [01:03<00:00, 98.53it/s]
```

fvollmer, 2019-04-26:
Sorry for the confusion. I somehow forgot `update-firmware`. I have no experience with Hyper-V. I flashed my robot from ubuntu (bare metal) and there were no firewall setting necessary.

The firmware update process creates a webserver to host the update package and provides the robot with a link to it. You have to make sure that the robot can access this file. 

If your VM is using NAT for network this won't work.

Philipp97714, 2019-04-26:
I can download the pkg file from any other device in my wifi using the link shown on the console. I don't know what to do now. Maybe xiaomi implemented checks in the update progress

onovy, 2019-04-27:
i already flashed new image with 1748 and 1768 on robot with current flasher, so it's not problem with firmware itself. it looks like it's problem with firewall on ubuntu side. Robot needs to connect "back" to your computer. It will probably not work with virtualization without network bridging, port forwarding, etc.

Philipp97714, 2019-05-04:
I don't think it's a firewall problem. I can get using the web browser of my pc.

rytilahti, 2019-05-04:
Have you checked out if the vacuum is doing a request properly (e.g., with wireshark)? Is it working with `mirobo`'s built-in update (it will also output the requests to console directly, so no need for wireshark there)?

rytilahti, 2019-05-04:
I'm pretty sure the problem is that there is no connectivity between your robot and your network (172.17.74.99). You can try `ip addr` to see your local IP addressesand pass `--ip <your wanted hosting address>` to the `update-firmware` command (see `mirobob update-firmware --help`), that should help.

Philipp97714, 2019-05-04:
You were right, the ip address is not reachable by the robot. I was able to get the file on my computer because it is in the same ip subnet as the vm (bridged) but the robo is not. I should have noticed it because the ip of the vm is not in my lan.

closed 2019-05-04 by Philipp97714