dgiese/dustcloud #228: Using USB port as a charging port?

HappyZ, 2019-05-31:
Hi there, 

If it's possible to use USB port to power up a small device, like raspberry pi? I tried to use OTG cable but it cannot draw power out of my S50.. Wonder if it's my OTG cable or just there's no power out through the port?

Another possible way is to route in parallel the battery and use a transformer to adjust the voltage and amps, but I'm trying to avoid the hardware mod..

Thanks.

dgiese, 2019-06-03:
Hi, a raspberry pi is not exactly a small power device, as it can draw up to 2.5A@5V. It basically depends also if you want it to be active all the time (also if the vacuum is powered off) or only if it is powered on.

HappyZ, 2019-06-03:
Or say raspberry pi one, which draws less power I believe (5V 1A). I only need it to be on when the vacuum is on. Do you have some suggestions? 

At last I can do is just use some portable batteries.. I guess it's doable if I want it to be on for just couple days.. but still, I wonder if that USB port has the power pin connected.. it could be my OTG cable problem.. but I don't understand as every USB port should have the VCC pin.. thanks in advance

HappyZ, 2019-06-04:
I just found some people can use OTG to draw power from the v1 xiaomi vacuum, however, it remains in question about S50.. I'll either get a different OTG to try, or test my OTG on my friend's v1 vacuum tmr..

HappyZ, 2019-06-05:
ok I can confirm it only works on v1 but not s5..

closed 2019-06-05 by HappyZ

betab0t, 2019-12-22:
@HappyZ How did you enable to power? I tried to connect an ESP32 to my gen 1 vacuum via both OTG & micro-USB cable but it doesn't seem to work...