dgiese/dustcloud #231: v1: using OTG to connect to other android devices

HappyZ, 2019-06-05:
hi, I got another question about this.

I can connect my phone to the v1 vacuum, and I can install the adb tools on the vacuum.
```
apt install android-tools-adb android-tools-fastboot
```

But when I run `adb devices`, I got `error: device offline`.

The android device works just fine on other linux systems. The system did recognize the device named:
```
>> lsusb
Bus 001 Device 011: ID 18d1:4ee2 Google Inc. Nexus 4 (debug)
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

But somehow adb cannot work. Any ideas?

dgiese, 2019-06-17:
Just guessing, but are kernel modules missing?

HappyZ, 2019-06-18:
yes I tried that, it was weird that it didn't work.. I didn't see much kernel messages regarding adb..

dgiese, 2019-12-22:
I think I figured out why: There is some switch that defines the role of the USB port. Even if you kill adbd, it might be still in adb mode.

closed 2019-12-22 by dgiese