dgiese/dustcloud #236: Viomi V2 vacuum support

MHOOO, 2019-07-05:
Hi,

I own a viomi vacuum, which according to the Internet is very similar to the xiaomi vacuums. At the very least it has ssh running on it and can be controller via the mi home app.

I wonder if trying out the firmware flashing is a good idea, or if I could irreversibly brick the device?

flatsiedatsie, 2019-07-16:
Where did you read that it can be controlled via the Xiaomi app? I read the opposite, that the app support is poor. https://smartrobotreviews.com/reviews/roborock-s6-vs-viomi-v2.html

adam-golab, 2019-07-17:
There are 2 versions of the Viomi V2 vacuum. The first one supports only Viomi App. And the second works only with the Mi Home App. If you are lucky and bought the vacuum with the Mijia logo on it, then you can use Mi Home App. 

Both versions have a micro USB port with SSH running. One can connect to it via ADB. I checked the first version and it runs originally on Tina Linux (based on OpenWRT), so I see one difference here, as Xiaomi Mi Robot works on Ubuntu (based on info from https://www.kaspersky.com/blog/xiaomi-mi-robot-hacked/20632/).

dgiese, 2019-07-23:
I will add support to the dustbuilder in the next days, which will however only work for the Mi Home compatible devices (I bought mine 2 months ago and its already Mi Home compatible). Unfortunately, the original ones do not contain the Mijia Cloud credentials, so there is no real official "upgrade" path for that devices. Due to the Squashfs image and the small NAND flash we are also limited in options like Valetudo. Will keep you updated.

hudarsono, 2019-08-23:
Any update on this?

t0kic, 2019-08-24:
Hi @adam-golab. How did you find out about the linux version on the robot? Mine also only supports the Viomi app and has the ports 22, 53, 6000 and 6008 open. 
Perhaps we can work with the xserver ports?

dgiese, 2019-09-03:
Access to Viomi is, compared to Rockrobo, more simple. I am preparing some tools for it.

t0kic, 2019-09-03:
That sounds great @dgiese
I also have an viomi v2 and I was thinking to send it back to the vendor due to the very bad viomi app and lack of dustcloud support. 
If I can help somehow or if you need me to test anything on my device please let me know!

dgiese, 2019-09-03:
If you have a Viomi-App only device, there seems to be an upgrade path to the Mi Home version. I am aware of at least one user who did that. He contacted Viomi via Facebook and they sent him some weird tool for it.

t0kic, 2019-09-03:
Yes, it is viomi app only device. Could you point me the way to that user or any other help how to change it to mi home version. 
Viomi FB is not responding to any of my messages so far

adam-golab, 2019-09-03:
@t0kic it usually takes around 5 days to get an answer from them on FB. So you have to be patient.

t0kic, 2019-09-03:
thanks @adam-golab. I asked them about one week ago, but will be more patient. 
As soon as I have any answer, I will let you know. 
There is a rumor, viomi will stop developing new firmware for the viomi-app only version.

adam-golab, 2019-09-03:
Yes, I've heard it too. Also, there were rumors that on August they upgrade all devices to Mi Home. But as September starts it only confirms that it was not exactly the truth. 

If you are waiting for an answer one week, I'd recommend asking them once again there. If it possible for you, write to them in their daytime. So around 10am - 4pm GMT+8. I was getting answers mostly around 4-5pm GMT+8. But this is not a general rule as a few times I've got a response on 12am GMT+8

t0kic, 2019-09-03:
Thanks for the hint @adam-golab! Will write them at that time. 
It is really very good device and it will be shame if I have to send it back.

uhrwerk, 2019-11-11:
Hi, are there any updates on this? Also got the Viomi App Version and it's so sad that the robot isn't supported anymore by Viomi and won't get updates while the Mi Home does get them. Seems like Viomi on Facebook isn't answering anymore.

BackSlasher, 2019-12-15:
@dgiese can I use the latest version with a Mi-using Viomi V2?
It's still not in the "supported devices" list, so I'm unsure.
Thanks!

t0kic, 2019-12-16:
Hi @dgiese !

Would it be possible to use dustcloud or valetudo on Viomi App Version of V2 with root access via ADB?

Thank you!

BrendanxP, 2020-01-19:
I sent messages to Viomi Facebook, starting from mid October 2019 until mid November. I just noticed they gave me a short reply on January 8 2020. That took almost three months to get a reply, let alone a fix to get Mi Home support.
I was wondering if anyone else has made any progress to get from Viomi App to Mi Home software? Or if anyone got new information about the topic.

uhrwerk, 2020-01-21:
> I sent messages to Viomi Facebook, starting from mid October 2019 until mid November. I just noticed they gave me a short reply on January 8 2020. That took almost three months to get a reply, let [...]

In this forum: https://www.roboter-forum.com/index.php?thread/34126-viomi-v-rvclm21b-erfahrungen-fragen-tipps-tricks/&pageNo=30
somebody who tries to find a way to upgrade to the mi version asked recently for data of a v2 mi version. Since then nothing new happened 

Also viomi saw my message on facebook but I'm left on read. I'm done with viomi.

bordino, 2020-04-11:
Hello 
I have a viomi v2 pro and I installed the update yesterday 3.5.8_0034 everything went well except that there is no longer washing in Y, it washes only in S.
How to put the wash back in Y, please?

[image]
[image]

Best regards

anywen, 2020-04-25:
> Hello
> I have a viomi v2 pro and I installed the update yesterday 3.5.8_0034 everything went well except that there is no longer washing in Y, it washes only in S.
> How to put the wash back in Y, please?
> 
> Firmware 3.5.8_0034
> [image]
> 
> Firmware 3.5.3_0030
> [image]
> 
> Best regards

I’ve got exactly the same problem with a viomi v2, did you found a workaround for this?
Best regards

tomiu, 2020-05-26:
In the application go to settings. There you will see "Mop or vacuum & mop mode". Click it and then chose Y-pattern

keynight, 2020-11-18:
https://github.com/keynight/viomi_v2  
Inside Viomi V-RVCLM21B with ABD/SSH access

closed 2021-12-12 by dgiese