dgiese/dustcloud #237: Is the Xiaomi Mijia 1S supported? (2019 upgraded version of the original)

flatsiedatsie, 2019-07-16:
I couldn't find anything through a search, so I thought I'd ask.

The Xiaomi Mijia 1S is a new iteration of the 'classic' Xiaomi robot vacuum. They've added a small camera so it can "recognise doors", and it should have some better mapping algorithms.

Does Dustcloud and its ecosystem support this device? If not, what are the odds that it will be supported? What would be required to explore that option?

More details: https://smartrobotreviews.com/reviews/xiaomi-mijia-1s-robot-vacuum-short-review.html
The device on Gearbest: https://www.gearbest.com/robot-vacuum/pp_009977063488.html

vmzhivetyev, 2019-07-23:
https://github.com/dgiese/dustcloud/issues/213#issuecomment-483629519

flatsiedatsie, 2019-07-24:
Thanks. That's unfortunate.

and-fed, 2021-01-20:
Any news? Maybe need to provide photos or other data?