dgiese/dustcloud #252: Gen 3 root ideas (at one's own risk)

lucidyan, 2019-11-27:
Dear @dgiese! As you describe in https://github.com/dgiese/dustcloud/issues/213#issuecomment-552096485, you have some ideas on how to root the latest Xiaomi vacuum cleaners (Mijia 1S, Roborock T6 / S6, Mijia 1C).

Can you share your thoughts on this or just tell us about your progress? Maybe you need [help from the community](https://github.com/dgiese/dustcloud/issues/213#issuecomment-552096485) or something else.

dgiese, 2019-11-28:
Hi,
the Mijia 1S is a totally different architecture in comparison to x6 and 1C. The approaches to root them are very different.

1S: Rockchip RK3326, with encrypted flash, uses Trustzone, Secure Boot and Android Verified boot
T6/S6: Allwinner R16, similar to S5/V1, however they changed the OS, signed configs, new encryption+signatures for formware
1C: not developed by Rockrobo, based of Allwinner MR133, uses Trustzone, very stripped down OS.

All of them cannot be updated from the LAN anymore. They locked down our usual way to root. Even if it would be still open, we cannot create a valid firmware due to the signatures. There are attacks for all 3 of them, but they are not trivial and I am still developing the tools for them...

oscar-rr, 2019-12-03:
Does the new Roborock S5 Max belong to this Gen3 too ?
Didnt find anything about root or flashing this device

dgiese, 2019-12-03:
S5 max (aka p5e) is similar to T6, however they changed part of the hardware and OS. For all the new devices after S5 (not max) the simple rooting methods are not working anymore :(

xsasx, 2019-12-04:
I have an 1S so if i could test or help in anyway please contact me

ivanwyc, 2019-12-15:
Is S4 gen 3? If it is, does it mean rooting S6 will be similar to S4?

dgiese, 2019-12-22:
Not sure about S4, I dont own one... I you have one and you disassembled it, send me an email.

novakduc, 2019-12-25:
Very irritated with my T6, I am ready to tear it down. @dgiese if I can do something to support the issue, please let me know.

daihashi, 2020-01-10:
> Not sure about S4, I dont own one... I you have one and you disassembled it, send me an email.

I have an S4 and am willing to tear it down. What information or pictures do you require?

randomodbuild, 2020-01-10:
I have an S6 and would also like to root it so I can use valetudo. I am mostly just trying to get the live map to show up in home assistant. I can do some testing but I probably cannot take many risky firmwares unless there is a safe way back because our houshold depends on it LOL. I hope I don't seem like one of those beggars asking for other people to make software for them for free. I'd be willing to donate a small bit of money for the cause though.

monster400, 2020-05-01:
I've got an S5 max i also will like rooted :)

lucidyan, 2020-05-14:
Dear @dgiese , is there any good news about rooting methods for our bricks? :)

jr-k, 2020-05-14:
> S5 max (aka p5e) is similar to T6, however they changed part of the hardware and OS. For all the new devices after S5 (not max) the simple rooting methods are not working anymore :(

you said "not max", can you be more precise ? The S5 max is rootable or not ?

monster400, 2020-06-06:
Hello.

I've taken som pictures of the s5 max. i hope it can help.

[](https://nextcloud.katea.dk/index.php/s/zrknAqpjksEeJ2T)

dgiese, 2020-07-02:
Check https://github.com/dgiese/dustcloud/issues/213 for information about S6/T6

sander1095, 2020-12-23:
Any updates about rooting my S5 max? I would like to use some custom sounds and all that

lucidyan, 2021-01-04:
@dgiese Hello, I recently found your POV video from MI1S camera. Could you please share your rooting progress and thoughts on whether this will be available in the future?
https://www.youtube.com/watch?v=JjYbkMARilY

astemos, 2021-04-16:
I'm also very interested in rooting M1S.
I'm willing to test it, however there's no guide as dgiese is not going to publish it.

evgenyh, 2021-06-12:
> 
> 
> I'm also very interested in rooting M1S.
> I'm willing to test it, however there's no guide as dgiese is not going to publish it.

the device is almost 2y.o. and what used to be 0dayz seems not to be categorized same 2 years later. so hopefully there might be a time to release a public way to root this device. crossing fingers. :)

dgiese, 2021-12-12:
For reference: See my DEFCON 29 presentation.

closed 2021-12-12 by dgiese

lucidyan, 2021-12-16:
Presentation PDF link [249MB]:

https://media.defcon.org/DEF%20CON%2029/DEF%20CON%2029%20presentations/Dennis%20Giese%20-%20Robots%20with%20lasers%20and%20cameras%20but%20no%20security-%20Liberating%20your%20vacuum%20from%20the%20cloud.pdf