dgiese/dustcloud #254: Cannot connect to vac anymore - cannot factory reset - no SSH access

NOTORIOUSVR, 2019-12-06:
I came home today and was unable to reach my Vac (S55 V2) on the network after almost 2 weeks of use without an issue. All the robot functions still worked if I used the buttons.

Resetting the wifi I am able to connect my laptop to it's internal wifi, but unable to bring up the web page or SSH into it.

My next attempt was a factory reset - which the voice prompt says will take 5 min... but nothing ever comes of it.

I'm not sure what to do next, the search also doesn't bring anything useful up.

HorizonKane, 2019-12-07:
I have the same issue.

Valetudo got lost few weeks ago already but I could easily reinstall it through robot control center.

Now it is gone, SSH isnot possible anymore. I resettet and added it to the Xiaomi Home App again, updated to 1886 like before and tried to flash my rooted firmware again, but when the flasher.py script trys to connect I only get:

```
(.venv) pi@berry:~ $ python3 flasher.py -f output/v11_001886.fullos.pkg
Flasher for Xiaomi Vacuum
Address is not set. Trying to discover.
Connecting to device 192.168.83.43...
Sending handshake to get token
Got error when receiving: timed out
error while checking device: No response from the device
```

NOTORIOUSVR, 2019-12-07:
Seeing your post, I realized I should try the latest Xiaomi app in the play store - after doing that I was able to connect to my Robot again but that seemed to be the only way the older app wouldn't find it at all.  

After getting it setup, I reverted the app back to ver. 5.4.54 so that I could capture the token and then proceeded to connect to it via RCC and uploaded again the other firmware.

Interestingly enough, the token was completely different - so whatever happened I am not sure if anyone needs me to do some digging I can but as of now I was able to get the FW back on again and it is working as it should.

HorizonKane, 2019-12-10:
How do you capture the token using the 5.4.54 version of the app and where to get that version?

My next try will now be to do a total reset, like suggested here: https://github.com/Hypfer/Valetudo/issues/206

NOTORIOUSVR, 2019-12-10:
> How do you capture the token using the 5.4.54 version of the app and where to get that version?
> 

I used this guide (has link the the APK):
https://selfhostedhome.com/zoned-cleaning-with-the-xiaomi-roborock-s5-robotic-vacuum/