dgiese/dustcloud #255: Can't connect to vaccuum

danieladr, 2020-01-02:
Hello, I am trying to convert a Chinese version of Roborock S5 to the International version, but I can't overcome this error.

Any ideas?

`python flasher.py -a 192.168.0.105 -t 48634542505250567243726c78324e77 -f output/v11_001720.fullos.pkg`

```
Flasher for Xiaomi Vacuum
Connecting to device 192.168.0.105...
error while checking device: {'code': -10002, 'message': 'access denied'}
```

I followed [this steps](https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto), and I only get stuck at the last bit.

jisaacs1207, 2020-01-14:
I’m having similar problems trying to do the same thing. Did you make any headway?

ratomatrix, 2020-02-09:
Hi,

I had the same issue, but found a workaround which worked for me.

The exception comes from flasher.py when calling vacuum.status() (line 157 in commit 61af137). Due to the exception, the execution would continue soon with an exit() call (line 166).
I just eliminated this exit() call, so that the script continues even if the status could not be retrieved. But in fact the flashing worked flawlessly for me afterwards.

Hope this helps.
Of course, do it at your own risk...

-ratomatrix