dgiese/dustcloud #258: Xiaomi Mijia Robot Vacuum STYJ02YM Bricked After .45 Update

Zabimaka, 2020-01-13:
Hello! Need help with severalbricked vac.
Several STYJ02YM Bricked after trying get new 45 update (update just freeze and after bricking the vac). Button's flashing, but no sign of life when pushing them. Factory reset, wi-fi reset didn't work. Turn off, turn on didn't work. Around 3 people from our community already have this issue
Is there any way to do factory reset via usb or flash factory firmware ? Xiaomi Community tell's nothing, only contact with point of purchase.
I have the same vac but i lucky was lucky with this update and create this topic to find solution for other people.

Thank you in advance!

kkszysiu, 2020-01-13:
@Zabimaka , also no sound when you turn it on/off?

Zabimaka, 2020-01-13:
Yes, just flashing buttons and it's all.

Zabimaka, 2020-01-13:
Thread on Mi Community

https://c.mi.com/thread-2796409-1-0.html?sources=1

пн, 13 янв. 2020 г., 23:58 Алексей Чернявский <bigmofoka@gmail.com>:

kkszysiu, 2020-01-13:
Yeah, I'm having Conga 3490 model (it's same model, VIOMI V2 clone), same, just flashing lights. I'm trying to resurrect this motherboard now. So far I see deleting *.txt.gz log files from /mnt/UDISK/log helped a bit and I'm hearing the startup sound but it's still being locked somewhere.

I had A LOT of logs there and I think that's why the upgrade failed, it was just not enough space but the device still tried to get upgraded.

Zabimaka, 2020-01-14:
Can you tell how you manage to connect to robot?
Any manual?

kkszysiu, 2020-01-14:
@Zabimaka, when you open battery container, on the bottom there is a micro USB connector:
[image]

There is also a micro USB connector on the bottom side, under rubber.

Just connect it to your PC, then use "adb" command, it is distributed with Android tools.
Tutorial here: https://www.xda-developers.com/install-adb-windows-macos-linux/
Not sure how outdated it is.

To connect into shell the command is:
```adb shell```

 After successful connection you should see:
```
BusyBox v1.24.1 () built-in shell (ash)

 _____  _              __     _
|_   _||_| ___  _ _   |  |   |_| ___  _ _  _ _
  | |   _ |   ||   |  |  |__ | ||   || | ||_'_|
  | |  | || | || _ |  |_____||_||_|_||___||_,_|
  |_|  |_||_|_||_|_|  Tina is Based on OpenWrt!
 ----------------------------------------------
 Tina Linux (Neptune, 57513AA3)
 ----------------------------------------------
```

That means you're in command line of a robot.

Zabimaka, 2020-01-14:
Thank you! It's a very useful information. Is there any way to take
firmware from working robot and flash it to bricked robot to make it work
properly? We are newbie in this questions, so sorry if question is dumb:(

kkszysiu, 2020-01-14:
I think there is not an easy way to do it just yet.
I think nobody did such flashing for this generation of robots, so it's kinda new grounds.

Zabimaka, 2020-01-14:
I guess without it or without deleting files that locking Robot it's no way to fix it (if Xiaomi maybe know's, but they keep silence every where so i guess they don't know to). No way to easy ctrl+c = ctrl+v i guess in this case:(
Can you please post any informartion if you get steps or success with revive robot ?

We would be very grateful!

dgiese, 2020-01-14:
Unfortuantly I have only the Viomi original model and not the Xiaomi specific one. I am not sure if the partition layout is the same for both or not.

However over that ADB you should be able to to read some of the boot logs and even might be able to run the processes manually.

Zabimaka, 2020-01-14:
Excuse me, how I can run process manually? Is there any special commands?
Sorry, like I said I am newbie in this.

falcon4fun, 2020-01-14:
@Zabimaka By the logic of *nix systems:
Thru dmsesg, logcat or cat-ing /var/log/ files.

kkszysiu, 2020-01-14:
Logcat is not supported.

I dumped the data from both bricked and working motherboards. Now I'm planning to compare both of them and check where is the difference.

Zabimaka, 2020-01-15:
I guess soon will be solution? That's great?

tovroman, 2020-01-16:
Is there any solution to rebuild a Styj02ym vacuum cleaner? I have the same problem = (

tovroman, 2020-01-21:
> Logcat is not supported.
> 
> I dumped the data from both bricked and working motherboards. Now I'm planning to compare both of them and check where is the difference.
Hello, have you got any repair solution?

dgiese, 2020-01-22:
Can you share the images with me? I can compare them against mine...
Send me a mail to : dgiese@mit.edu

tovroman, 2020-01-22:
Sorry, I'm new to this business, what photo do you need? If any ADB commands, please describe in detail.

tovroman, 2020-01-22:
I can give you remote access to my computer, can you execute the commands you need, is it convenient for you?

kkszysiu, 2020-01-22:
Hey guys,
I haven't checked it recently. I will try to look on it over weekend if I found some time.

volodya13180, 2020-01-23:
Rooting the Xiaomi STYJ02YM (viomi-v7) Vacuum Robot https://itooktheredpill.irgendwo.org/2020/rooting-xiaomi-vacuum-robot/

tovroman, 2020-01-25:
This method will help if my vacuum cleaner blinks slowly but does not respond to pressing and cannot work?

szerwi, 2020-01-30:
I have the same issue (thread on Mi community is mine).
I've tried connecting the vacuum to the PC, but after connecting the vacuum there is no sound in Windows (it is not detecting any new device). Is it still possible to connect to the vacuum by adb software or my USB port in the vacuum is broken?

minscoop, 2020-02-02:
Hi all,

Any progress or solution on this?
I am really frustrated to Xiomi. How can I make them move into this error?

chanwooo, 2020-02-06:
My vac could not connect to adb
adb does not recognize the device
  Please tip if you are successful

projek01, 2020-02-08:
Hi guys, my vacuum decided to fail a firmware update last night as well. Being an imported machine from China, claiming warranty would be impossible. 

If you guys are able to figure this out, please please share with us. I would be more than happy to buy someone a few beers for their hard work.

Ninienabou, 2020-02-08:
I Ve got the same issue, anyone successes in hard reseting the robot?

projek01, 2020-02-09:
After trying to get ADB access to my vacuum, I have determined that mine is stuck in a boot loop. It will connect to my USB for about 3 seconds, disconnect for about 5 seconds and then reboot. Definitely not enough time for Windows to recognize and install drivers let alone access any files. 

I think the only fix for this problem is either a new motherboard or if someone can flash the firmware via JTAG. There is a 4 pin header right next to the USB that I bet is for that purpose. That is really above my skill set but maybe we can pitch in and set up a bounty for someone smart to help us.

eldeeone, 2020-02-11:
Exact same issue here, several site also speaking about it :
in French here [https://forum.hardware.fr/hfr/electroniquedomotiquediy/domotique_maisonconnectee/styj02ym-aspirateur-laveur-sujet_1557_15.htm#t291037](url)
in English [https://c.mi.com/thread-2796409-2-1.html](url) and [https://www.roboter-forum.com/index.php?thread/40257-robot-aspirateur-xiaomi-styj02ym-problème-mise-à-jour-fw/&pageNo=4#post543864](url)

Ninienabou, 2020-02-11:
One more thread talking about the same update problem: https://www.dealabs.com/discussions/xiaomi-mijia-styj02ym-p2019-hack-1717753

projek01, 2020-02-12:
Something I found on the French thread that I tried:
1. Press and hold power and "home" button while plugging in USB = LED becomes solid white.
2. Press and hold power and "home" button while plugging in USB but letting go of only power button when the LED turns orange = LED becomes solid orange. 

Both methods don't result in a stable USB connection for me. Only connects for about 2 seconds and the disconnects again and again.

I am hoping to get into FEL mode that the Aliwinner processor supports. This will allow for flashing the firmware over USB. In reality, I don't really know what I am doing. Just trying random stuff.

kkszysiu, 2020-02-12:
I will love to help you guys, my RobotApp is starting already and I'm in process where it checks for connected sensors, motors etc., but the issue is that I have motherboard only. The other robot was simply fixed by changing the mobo.

I'm searching for local auctions to get bricked or damaged robot - maybe that way I will be able to help you guys.

@projek01 , the reason why your USB is starting only for 2 seconds, then turn off is because of RobotApp, it means it's turning on and during startup of this app it disables adb access.
You can easily turn off the RobotApp and then you will gain SSH access and you'll be able to access logs.

projek01, 2020-02-12:
@kkszysiu 
Thank you for your help and reply. So that makes sense and my RobotApp is likely stuck in a boot loop.

I am looking at the root tutorial and it looks like he is using either OSX or a linux distribution. I have never used either but I am downloading Ubuntu right now. Everything in that tutorial makes sense to me except for the following:

> To fix this, we create our own version of this file:
> cat >adb_shell <<EOF
> #!/bin/sh
> export ENV='/etc/adb_profile'
> exec /bin/sh $@
> EOF
> chmod +x adb_shell

Is this command run line by line or can it be copied and pasted into the terminal in one block. Sorry, I am a total linux command line NOOB.

falcon4fun, 2020-02-12:
@projek01 
Either.
cat command inserts to adb_shell text betwee <<EOF and EOF
So it will wait for "EOF"

Example using line by line:
[](https://i.imgur.com/iRjnOQB.png)

projek01, 2020-02-12:
@falcon4fun 
Thank you for that explanation. It's nice to have the process broken down into pieces. 

Reminds me of a good calculus professor where you understand how it was derived versus a crap professor in school and you are forced to memorize by brute force.

projek01, 2020-02-13:
My vacuum seems to be in a deeper hole compared to kkszysiu. According to my USB device and vendor ID. Mine is actually going into FEL mode for 2 seconds and then rebooting. I made this assumption based on the following:

> Device=ff.ff
[ 2733.131960] usb 3-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2735.917395] usb 3-4: USB disconnect, device number 109
[ 2749.007512] usb 3-4: new high-speed USB device number 110 using xhci_hcd
[ 2749.155791] usb 3-4: New USB device found, idVendor=1f3a, idProduct=efe8, bcdDevice=ff.ff
[ 2749.155796] usb 3-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2751.945094] usb 3-4: USB disconnect, device number 110
> 

Just as expected, the global Xiaomi support just brushed me off.

szerwi, 2020-02-13:
@kkszysiu I'm from Poland and I have bricked motherboard, but for now I want to repair my vacuum, not sell it. Do you have any working motherboards that I can buy?

eldeeone, 2020-02-20:
Anyone got news from vendor/Xiaomi/Tierce to solve the issue for its vacuum ?

Ninienabou, 2020-02-20:
Xiaomi answers me to contact local customers service if product available (but this robot is only for China so customers service is unavailable for me in Europe) and seller doesn't answer (Aliexpress order, My robot hub seller).
I have another one of these robot working well in .42 software. But I don t know if I can backup and install software from my working robot into the bricking one

eldeeone, 2020-02-22:
Good news my side, the vacuum that was on the charging base blinking for like 2 weeks seems to be back to life... Yesterday it stops blinking and I was able to reset wifi then gain access to it again. I did nothing beside keeping it on the charging base and nothing happen special (neither updated the app on my iPhone)

szerwi, 2020-02-22:
@eldeeone Interesting... Why did you have to reset WiFi in your vacuum? Did you change password to WiFi or the vacuum just didn't want to start working before the reset?

ercintorun, 2020-03-12:
İ do have the same problem also. İ can try to help connecting ADB and taking logs.

minscoop, 2020-03-12:
> Good news my side, the vacuum that was on the charging base blinking for like 2 weeks seems to be back to life... Yesterday it stops blinking and I was able to reset wifi then gain access to it [...]

Was the blinking light red or white?

eldeeone, 2020-03-12:
> @eldeeone Interesting... Why did you have to reset WiFi in your vacuum? Did you change password to WiFi or the vacuum just didn't want to start working before the reset?

Keeps showing offline on the app that’s why I reset wifi

eldeeone, 2020-03-12:
> Was the blinking light red or white?

Was slowly blinking white

redman6, 2020-03-21:
@eldeeone @ szerwi @Ninienabou @projek01 and others.... Has anyone been able to fix their robots ?

Ninienabou, 2020-03-22:
> @eldeeone @szerwi @Ninienabou @projek01 and others.... Has anyone been able to fix their robots ?

No, it seems to be no solution, it seems also that. 45 update is not the only that bricked the robot, others reports bricked robot after 46 update.

projek01, 2020-03-24:
no luck here. also still bricked. also dont see any motherboards for sale still.

mLupine, 2020-03-28:
I've managed to do quite a lot of progress on getting my bricked device back to life. I need some help from someone with a working and rooted robot, though, to be able to finish my work. See my comment (and the entire thread) [here](https://github.com/rumpeltux/python-miio/issues/1#issuecomment-605454925) , and let me know if you can help!

jolenders, 2020-03-28:
I hope it works they ask 180€ for a new motherboard 🤦‍♂️

szerwi, 2020-03-28:
@mLupine great, I hope you will find someone with working robot. I guess @kkszysiu has one.
@jolenders where did you find spare motherboard?

jolenders, 2020-03-28:
True my seller on ali express

ercintorun, 2020-04-16:
@mLupine Thank you very much for finding and explaining the whole process. I also recovered my bricked vacuum with your method. Thanks.

projek01, 2020-04-20:
Also unbricked my vacuum using mLupine's guide. At first I just randomly shorted 2 pins on my eMMC but I had a scare when a pin started to glow a little. Instead, try to connect pin 8 and 13 specifically. Thanks so much!!

zoro956, 2020-05-13:
> Also unbricked my vacuum using mLupine's guide. At first I just randomly shorted 2 pins on my eMMC but I had a scare when a pin started to glow a little. Instead, try to connect pin 8 and 13 specifically. Thanks so much!!

hi, I can find nothing about shorting pins in the mLupine's guide. Could you explain this? Thanks.

JanHavlicek, 2020-08-06:
Does anybody have the firmware file? All links mentioned before in all articles aren't working anymore. Thank you very much

dgiese, 2020-08-06:
Which one you need?

JanHavlicek, 2020-08-06:
Already got one in different discussion, it’s here - https://gofile.io/d/T7Pcjz , thank you :)

ovzap, 2020-10-22:
Hello,
During the upgrade process something went wrong and now NAND has several bad blocks. I've resoldered old NAND with new one and now I need to write the firmware to it.
Could you advice me where I could get the dump of NAND with firmware for STYJ02YM?

ciffer, 2021-01-08:
Hi,
I am also looking for a firmware file (actually I own a Viomi V2 pro, but the mobo is the same apparently). I have a friend with the same model so if you can instruct me on extracting the url/file from a working unit I might be able to do it fairly easily.

[3 more replies]

closed 2021-12-12 by dgiese