dgiese/dustcloud #261: audio_generator lang files out of date for new gen2 firmware

fredless, 2020-01-26:
My S5 running stock 3.5.7_002008 recently said "Positioning, please wait."  This prompt is not listed in  [audio_en.csv](../blob/master/devices/xiaomi.vacuum/audio_generator/language/audio_en.csv) so I am assuming the prompt packages are out of date for this latest gen2 firmware.

fredless, 2020-01-26:
Preliminary digging, comparing the built-in en files of v1886 vs. v2008 firmwares, there are indeed 16 new sound files:

```bash
$ sounds1886="1886image/opt/rockrobo/resources/sounds/en"                          
$ sounds2008="2008image/opt/rockrobo/resources/sounds/en"
                          
$ ls -1 $sounds1886 | wc -l
79

$ ls -1 $sounds2008 | wc -l
95

$ find "$sounds1886/" "$sounds1886/" "$sounds2008/" -printf '%P\n' | sort | uniq -u
ding.wav
map_restore.wav
positioning.wav
restart_clean_fromdock.wav
restart_clean_nodock.wav
resume_backtodock.wav
resume_room.wav
resume_spot.wav
room_complete.wav
room_failed.wav
room_partialdone.wav
room.wav
saving_map.wav
stop_clean.wav
stop_room.wav
zone_partialdone.wav
```

fredless, 2020-01-27:
Submitted pull #262 to _partially_ address, includes updated english lang transcription based on files included with 3.5.7_002008.

Note that in addition to the new files above, there were significant changes to the existing prompts.  Far fewer uses of the word "please", "the cleanup" has been largely replaced with "cleaning" or "clean", and "vacuum" has been replaced with "main unit" in most places (while I think the new prompts are an improvement overall, I think this last "main unit" bit moved the friendly needle in the wrong direction).

Dfm1011, 2020-03-30:
Fredless, I’m looking for a download of the new English voice pack. Do you happen to have it?

fredless, 2020-03-30:
> looking for a download of the new English voice pack

@Dfm1011 if by "pack" you mean the contents of the `/opt/rockrobo/resources/sounds` folder from the 3.5.7._002008 firmware image, then yeah, I've got those.  It's 35.75MB zipped.  Need it?

Dfm1011, 2020-03-30:
Yes!! Could you please provide a download link or something lol! Thanks

fredless, 2020-03-30:
@Dfm1011 will leave it up for a few days and then delete it again, as it's cruft

Dfm1011, 2020-03-30:
Thank you SO much! Seriously!