dgiese/dustcloud #263: Roborock Gen 2 - No SSH after Update 002008

asch8505, 2020-02-05:
Since the latest update (room cleaning update) I am not able to open an SSH connection to the device. Miio is still connecting and it is possible to flash the same firmware again and again.

I guess they included some kind of firewall rule to the roborock to block SSH. Nmap tells that the port 22 is open, but filtered.

Is somebody having the same issue with the new update and knows how to fix it?

dugite-code, 2020-02-06:
If memory serves from the Valetudo repo the latest firmware has been locked down more. In fact they advise:

>PSA: DO NOT UPDATE YOUR GEN2 TO 1910 & GEN1 TO 4004

I couldn't find the specific details again but doing a [factory reset](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset) to down-grade is the best option at the moment.

github4pbp, 2020-03-08:
In the new S5 FW: 3.5.7_002008 they might changed the sshd with dropbear and added some kind of firewall. I have noticed dropbear presence in /usr/sbin/dropbear and /etc/dropbear/ where the keys are placed. Furthermore it is configured to respawn in the newly added inittab as follows: "::respawn:/usr/sbin/dropbear -F" Still cannot figured out what filters port 22 as line "iptables -I INPUT -j DROP -p tcp --dport 22" in rrwatchdoge.conf is commented upon CFW build...

FeatureExpert, 2020-04-05:
Just remove iptables binary from FW image and you will be able to log in with proper root password

t773, 2020-04-20:
I saw somewhere that openssh and many other stuffs have been taken off from 2008 fw so the size was reduced from 1xx MB to 6x MB. Is it possible to put ssh in the modified fw?

git4pbp, 2020-07-25:
From firmware 3.5.7_002008 they started using BusyBox v1.24.1 (2019-11-15 15:22:50 CST) built-in shell (ash) and openssh is replaced by dropbear. The replacement of dropbear with openssh should be possible, but bear in mind that port 22 seems to be filtered via iptables on binary level in /opt/rockrobo/watchdog/WatchDoge. Therefore it doesn't matter whether you comment _iptables -I INPUT -j DROP -p tcp --dport 22_ in rrwatchdoge.conf or not.

Hex editing that binary as shown below you should able to do ssh with your extracted root password from vinda

[image]
[image]

[root@rockrobo ~]#  netstat -tanp | grep :22
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      346/dropbear
tcp        0      0 192.168.1.11:22        192.168.1.239:62028    ESTABLISHED 1184/dropbear
tcp        0      0 :::22                   :::*                    LISTEN      346/dropbear

closed 2021-12-12 by dgiese