dgiese/dustcloud #267: Elaborate: Is there any server component needed for rooting S5?

LordMike, 2020-03-11:
Hi,

I'm very interested in rooting my vacuum cleaner, but I'm really confused as to whether a server component is needed. The docs don't say that there is, but when I see texts like this:

> where 192.168.42.42 is the IP of your local instance and the port number on which server.py runs (can technically run on a Raspberry pi).
> 
> https://github.com/dgiese/dustcloud/wiki/Preparations-on-the-Vacuum-Robots

And I [see this](https://github.com/JackGruber/docker_dustcloud), which comes up with a little webserver that has a list of devices .. (I dunno yet, if each device allows control or whatnot).

It was my impression that the Valetudo app/webpage was served from the device itself, and not somewhere else.. 

... so what is this server needed for?

dugite-code, 2020-03-13:
The Dustcloud server is a implementation example for intercepting data from the vacuum going to Xiaomi and was intended as a template for other developers to work off of and not a final product. The Live map is an old way of decoding the map data on the vacuum.

Valetudo has a newer method of extracting the map and yes mostly runs on the vacuum, however map decoding is done in the web browser. So if you want the live map in anything else, like homeassistant, you need to use [ICantBelieveItsNotValetudo](https://github.com/Hypfer/ICantBelieveItsNotValetudo/)

 As you wish to use Valetudo it's a good idea to follow the installation instructions on [its repository](https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions).

How the rooting method works:

1. You patch a copy of the original firmware to enable ssh (and possibly add Valetudo).
2. You run a flashing script that:
    * Uses python-miio to handshake and connect to the vacuum
    * Passes a url to the vacuum to download the patched firmware
    * Starts a webserver to serve up said firmware to the vacuum
3. Once the vacuum has installed the patched firmware it now has ssh (and possibly Valetudo if you added that).

LordMike, 2020-03-13:
Ah. Great. It was as I hoped - I had read about the live map thingy being remote - that suits me fine.

Great work, thanks :)