dgiese/dustcloud #269: flasher.py: AttributeError: 'Vacuum' object has no attribute 'do_discover'

noscript, 2020-04-12:
Using the latest `python-miio` (at this moment `0.5.0.1`) fails with:

```
$ python firmwarebuilder_0.3/flasher.py -f output/v11_001856.fullos.pkg
Flasher for Xiaomi Vacuum
Address is not set. Trying to discover.
Connecting to device 192.168.8.1...
Sending handshake to get token
Traceback (most recent call last):
  File "firmwarebuilder_0.3/flasher.py", line 210, in <module>
    main()
  File "firmwarebuilder_0.3/flasher.py", line 151, in main
    m = vacuum.do_discover()
AttributeError: 'Vacuum' object has no attribute 'do_discover'
zsh: exit 1     python firmwarebuilder_0.3/flasher.py -f output/v11_001856.fullos.pkg
```

At least `do_discover()` has been renamed to `send_handshake()`.

Workaround for now is to install the older version:
```
$ pip install -Iv python-miio=0.4.8
```

vacuumfan, 2020-04-14:
Thank you.
Was struggling with that.
Everything worked great for a year.
Firmware update. Everything broken. tried reflashing. Didn't work. 

'Vacuum' object has no attribute 'do_discover'

This sounds like a solution.
Will try again tomorrow!

Stupid 3.3.9_001886....:-(

vacuumfan1, 2020-04-17:
It worked, but it was: 

pip**3** install python-miio=**=**0.4.8 

for me! (Ubuntu 18.04).

The line in post#1 gave back: ...python version 2.x but you need >3.x...
although python 2 and 3 was installed before.

But thanks a lot for the solution.
Everything back to normal now.
Hope it doesn't self destroy again...