dgiese/dustcloud #277: Xiaomi Gen1 firmware 3.5.4_004010 error

0x00xx, 2020-07-09:
I am trying to flash dustcloud on my newly bought robot, but I am encountering an error:

```
mirobo --ip 10.0.0.49 --token XXX6a6e59354769386837 info
rockrobo.vacuum.v1 v3.5.4_004010 (64:90:C1:01:0C:33) @ XXX - token: XXXa6e59354769386837

python3 flasher.py -a 10.0.0.49 -t XXX6a6e59354769386837 -f output/v11_004018.12edfghb75.fullos.pkg
Flasher for Xiaomi Vacuum
Connecting to device 10.0.0.49...
Starting local http server...
Serving http server at 10.0.0.46:36623
Sending ota command with parameters: {"mode": "normal", "install": "1", "app_url": "http://10.0.0.46:36623/v11_004018.12edfghb75.fullos.pkg", "file_md5": "8155f7811f7c8bd5e04881033c1b8847", "proc": "dnld install"}
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/miio/protocol.py", line 193, in _decode
    return json.loads(decoded)
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 355, in raw_decode
raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "flasher.py", line 210, in <module>
    main()
  File "flasher.py", line 188, in main
    r = vacuum.send('miIO.ota', ota_params)
  File "/usr/local/lib/python3.8/dist-packages/miio/device.py", line 146, in send
    return self._protocol.send(
  File "/usr/local/lib/python3.8/dist-packages/miio/miioprotocol.py", line 183, in send
    m = Message.parse(data, token=self.token)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 1981, in _parse
    subobj = sc._parsereport(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 2439, in _parse
    return self.subcon._parsereport(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 3987, in _parse
    obj = self.subcon._parsereport(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 4265, in _parse
    obj = self.subcon._parsereport(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "/usr/local/lib/python3.8/dist-packages/construct/core.py", line 698, in _parse
    return self._decode(obj, context, path)
  File "/usr/local/lib/python3.8/dist-packages/miio/protocol.py", line 199, in _decode
    raise PayloadDecodeException(
miio.exceptions.PayloadDecodeException: Unable to parse message payload
```

What is the cause of this error? Am I doing something wrong or am I not able to root this firmware?

Thanks

benhab-i, 2020-07-10:
Hi, my god am I looking for someone with the same problem and finally here you are !
As you said, I didn't saw anything about Gen1 firmware 3.5.4_004010. Here is my stacktrace
`(.venv) i.benhabib@LGOP3589:~/Téléchargements/vaccum/firmware$ ./flasher.py -f output/v11_003468.fullos.pkg 
Flasher for Xiaomi Vacuum
Address is not set. Trying to discover.
Connecting to device 192.168.8.1...
Sending handshake to get token
Starting local http server...
Serving http server at 192.168.8.66:36137
Sending ota command with parameters: {"mode": "normal", "install": "1", "app_url": "http://192.168.8.66:36137/v11_003468.fullos.pkg", "file_md5": "320a288d7b15e67c4769f2d3fa898f48", "proc": "dnld install"}
unable to parse json '': Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "./flasher.py", line 210, in <module>
    main()
  File "./flasher.py", line 188, in main
    r = vacuum.send('miIO.ota', ota_params)
  File "/home/i.benhabib@gop.link/Téléchargements/vaccum/firmware/.venv/lib/python3.8/site-packages/miio/device.py", line 278, in send
    self.__id = m.data.value["id"]
TypeError: 'NoneType' object is not subscriptable`

It's look like the parser is not working anymore on our firmware. I already tried a Factory Reset but it keep build the 3.5.4_004010 version.

0x00xx, 2020-07-10:
My guess is that the firmware prevents unofficial updates since maybe 3.5.0. But I did not find any confirmation on the internet.

So I hope it is not the case but who knows.

dgiese, 2020-07-10:
Hey, if I remember correctly, version "3.3.9" is the last one which supports local OTA. Thats at least the version we package with custom firmwares, as it is known to work.

benhab-i, 2020-07-10:
Hi @dgiese, thank you for your answer, so no way to hope a rooted firmware on 3.5.x ?

dgiese, 2020-07-10:
Oh, did not say that. Its just not as trivial any more as it was with local updates. Depending on the age of your robot, you might downgrade the firmware to its initial one (assuming its based on 3.3.9). Otherwise you need to use an alternative way, likely by disassembling the device. You can technically use the same process as for the S6/T6 (check out my Youtube channel for that).
You can also contact me over Telegram to discuss potential shortcuts (which may or may not apply).

0x00xx, 2020-07-10:
Well the robot comes shipped with 3.5.4_004010, I verified that through a factory reset. So I guess the only way to root it is to disassemble it. Gonna check out your youtube

dgiese, 2020-07-10:
I checked my magic firmware list here: https://dontvacuum.me/robotinfo/firmwares/roborock.vacuum.s5.html

I am at the moment not sure if 3.5.4 was already locked down for local OTA. But it could be that the SysUpgrade enforced download over SSL with special headers. https://builder.dontvacuum.me creates that headers. You might give it a try by building firmware and triggering a firmware update using the direct URL from the server.

0x00xx, 2020-07-10:
Definitely gonna check it out and give feedback afterwards

0x00xx, 2020-07-10:
Just a follow up question: When creating an image with https://builder.dontvacuum.me/ do I flash it with flasher.py?

0x00xx, 2020-07-10:
Change the ota_params of flasher.py like this:
```
ota_params = {
        'mode': 'normal',
        'install': '1',
        'app_url': 'https://builder.dontvacuum.me/jobs/j5f086bf3c34ed/08f6cbf41ea60e02219fb9e7fef308ea/v11_004018.pkg',#'http://{ip}:{port}/{fw}'.format(ip=local_ip, port=http_port, fw=os.path.basename(firmware)),
        'file_md5': '08f6cbf41ea60e02219fb9e7fef308ea',#md5(firmware),
        'proc': 'dnld install'
    }
```

but this still results in the same error :/

dgiese, 2020-07-10:
:( then you need to disassemble it and use UART. You can find the testpoints here: https://dontvacuum.me/teardowns/

Surfer2010, 2020-09-05:
I have the same problem.
new bought Roborock Mi (Gen1) and the factory reset keeps giving me a firmware of 3.5.4_004010

Is there ANY way that this will be doable like the versions before? OTA so i do not need to disasemble the device? Would be really great to have a tutorial on how to still do it without hardware access to it.
Thanks for your support, the older one i have works awesome with valetudo-dustcloud and i love it.

So please please help us to get this new firmware "cracked" also :-)
Thousand thanks

hypePG, 2020-09-12:
hey, i got the same problem. the robot comes with a firmware which is to new on both partitions, meaning a reset to an earlier firmware wont work. @dgiese is there a tutorial for flashing firmware via uart? thx in advance

LoggeL, 2020-11-04:
> 
> 
> hey, i got the same problem. the robot comes with a firmware which is to new on both partitions, meaning a reset to an earlier firmware wont work. @dgiese is there a tutorial for flashing firmware via uart? thx in advance

https://www.youtube.com/playlist?list=PL9PoaNtZCJRZc61c792VCr_I6jQK_IdSb

Here's the only one I could find

dgiese, 2021-12-12:
for reference: https://builder.dontvacuum.me/howtos/x6cheatsheet.txt

closed 2021-12-12 by dgiese