dgiese/dustcloud #278: [Request] tanos_en.pkg

haroldboom, 2020-07-13:
Hi all,

Does anyone have the tanos_en.pkg voice file? In the firmware section the preview for the English voice pack is there but not the actual file. Is there anyway to extract it from the firmware or direct download the voice pack? Thanks

dgiese, 2020-07-13:
English is already in the firmware. You can just switch to it.

haroldboom, 2020-07-13:
> English is already in the firmware. You can just switch to it.

Hi dgiese, I have the T6 so it doesn't have an English voice pack. 

Btw Ty for your work on rooting the S6/T6 I think I will just do that and install Valetudo, does that have an English voice pack as default?

dgiese, 2020-07-13:
It does, as the firmware of the T6 and S6 is exactly the same. Look under /opt/rockrobo/ressources/sounds. The folder contains the english, chinese and taiwanese version of the sounds.

haroldboom, 2020-07-13:
> It does, as the firmware of the T6 and S6 is exactly the same. Look under /opt/rockrobo/ressources/sounds. The folder contains the english, chinese and taiwanese version of the sounds.

ok awesome, so one last question if I decide not to use Valetudo and use your firmware without signatures. If I want to change my region to Australia I edit device.conf and change model=roborock.vacuum.t6 to model=roborock.vacuum.s6 

and edit roborock.conf to;

language=en
name=custom_A.03.0040_CE
bom=A.03.0040
location=au
wifiplan=
timezone=Europe/Berlin <- dont think this matters
logserver=awsde0.fds.api.xiaomi.com
featureset=1
cpuid=xxxxxxxxxxxxxxxxxxxx 

Thanks again for all your work, you are an absolute genius!

dgiese, 2020-07-13:
No, as said, the firmware and hardware of t6 and s6 is the same. The only difference are these 2 config files.

haroldboom, 2020-07-13:
great ty again!

closed 2020-07-13 by haroldboom