dgiese/dustcloud #279: File System Read-Only - Removing T6 region lock

haroldboom, 2020-07-16:
Hi DG,

I have rooted my roborock T6 following your instructions and installed your non-valetudo firmware because I just want to change my T6 to an S6.

However, when I go to edit the device.conf and roborock.conf in /mnt/default I get the following error in WinSCP and vi;
scp: /mnt/default/device.conf: Read-only file system

Did I miss something? I was following https://www.roboter-forum.com/index.php?thread/45980-region-geo-blocking-entfernen-t6-t4/ but translated to english.

Thanks

dgiese, 2020-07-16:
You need to remount the default partition as rewritable:
"mount -o remount,rw /mnt/default"
Make backups of the files you change!

haroldboom, 2020-07-16:
awesome that worked and shes speaking english! Thanks for all your hardwork mate

closed 2020-07-16 by haroldboom