dgiese/dustcloud #280: Convert viomi v-8 (EU) to viomi v-7(CN)

xxTudorxx, 2020-07-16:
So I have a Xiaomi Vacuum Mop P(Pro) from Europe, and It does not get the latest firmware updates like the viomi v-7 from China, also I cannot use it with the chinese mainland version in xiaomi home. Is there a way to convert it to the viomi v-7 variant and use it as if it was bought from the chinese mainland?
Also Xiaomi Mi Vacuum Mop P(EU) = STYTJ02YM = viomi v-8
And Xiaomi mi vacuum Mop P(CN) = STYJ02YM = viomi v-7
I have a idea to cancel the noise made by the vacuum by recording the noise, inverting it and replacing the sound that plays when the vacuum is started with the inverted track of the noise
Also I am very bad with the adb shell or whatever shell is used to controll the vacuum, if someone with experience and knowledge could tell me what commands to use and in wich order it will be very helpful. 
Thanks in advance!
Also, english is my second language, so sorry for any typing mystake. 
If I succed at canceling the noise made by the vacuum i will put here a video of it.

LZsolt75, 2021-01-16:
Are you sure about that STYTJ02YM is Viomi v8? I have bought one from Aliexpress and the type on the bottom shows that mine is also STYTJ02YM but in Mi Home app it show that the model is viomi.vacuum.v7. The voice is chineese and I cannot change it  to english or change is to whatever.
[image]

dgiese, 2021-12-12:
For reference: There is now a conversion firmware from v7/v8 to v6 on builder.dontvacuum.me

closed 2021-12-12 by dgiese

zhorvath83, 2022-11-29:
@dgiese Could you help me the firmware (for Viomi v-6) generated at builder.dontvacuum.me how can be flashed to the STYTJ02YM (v-8)?

I tried it with PhoenixSuit_V1.10 (it connects successfully to the robot), but doesen't accepts the unpacked img-s originated from the generated firmware.

Also copied it to the Robot by netcat, but I don't know how to flash them to the relevant partitions.

petec999, 2023-01-30:
@zhorvath83 , @dgiese what are benefits of loading v8 to v6? I have v8 EU, unfortunately I don't have multimapping in v8, what is available for v8 CN..., and that would be good for me (uploading of v8 CN)...

TheWeeezel, 2023-02-04:
> @dgiese Could you help me the firmware (for Viomi v-6) generated at builder.dontvacuum.me how can be flashed to the STYTJ02YM (v-8)?
>
> I tried it with PhoenixSuit_V1.10 (it connects successfully to [...]

I would also like to know. I have no idea how to do this.