dgiese/dustcloud #286: Which firmware should I get for STYTJ02YM?

ooii, 2020-09-02:
Hi,

I'm about to root my STYTJ02YM vacuum following the guide [here](https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto#instructions-on-linux) and I'm at step where I should download the firmware following [this guide](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Firmware). At this stage, I'm confused which firmware I should download.
My vacuum is a STYTJ02YM model v8 running currently firmware 3.5.3_0045, can you please tell me which file should I choose? Thanks.

dgiese, 2020-09-02:
Its not supported *yet*. I don't have that particular device to verify the rooting method and firmware. 
Check https://dontvacuum.me/robotinfo/ for further information. I will update that page when the tooling becomes available.

zvldz, 2020-09-02:
> I'm about to root my STYTJ02YM vacuum

https://github.com/rumpeltux/viomi-rooting

dgiese, 2020-09-02:
Not sure if that still works with that firmware version. At some point they disabled adbd and downgrading firmware.

ooii, 2020-09-02:
> > I'm about to root my STYTJ02YM vacuum
> 
> https://github.com/rumpeltux/viomi-rooting

I tried that and did not succeed (cf. https://github.com/rumpeltux/viomi-rooting/issues/7)

ooii, 2020-09-02:
> Its not supported _yet_. I don't have that particular device to verify the rooting method and firmware.
> Check https://dontvacuum.me/robotinfo/ for further information. I will update that page when the tooling becomes available.

Thanks. I can test and report to you if you want.

dgiese, 2020-09-02:
I prefer to test myself, as for that kind of devices a recovery is not possible. Last time I tested, I bricked my S5 Max. In theory valetudo supports the v6,v7,v8. I personally have a v6, but I did not really do anything with that after I rooted it in July 2019.  I let the developer of Valetudo (Sören) use it to test and port Valetudo. The v7 and v8 models do not have high priority for me at the moment. I potentially know how to package the payloads for them, but that requires some testing with a real device (=rooted device).

ooii, 2020-09-02:
I understand. Thanks for the explanation.
If I get it well, it is possible to root my vacuum and run Valetudo on it, right?

dgiese, 2020-09-02:
At some point yes. But it requires a customized firmware.

0k00l, 2024-12-10:
Is it supported already? On https://builder.dontvacuum.me/_viomi_v7.html i see 'V7,V8 conversion to V6 (V6 ver 0041, 11/2020) ' option. I manage to run dropbear(ssh) on my STYTJ02YM(viomi.vacuum.v8) and i have root access.