dgiese/dustcloud #299: Imagebuilder.sh public key error

AdrianFajt, 2021-01-21:
Hi all,

I am facing the following issue with the imagebuilder.sh:

I get the error:

Public key  doesn't exist or is not readable

Even though the keyfile is present in the firmware builder directory.

[image]

I am using the command:
./imagebuilder.sh -f v11_001886.fullos.pkg -s english.pkg -k id_rsa.pub -t Europe/Berlin --disable-firmware-updates --disable-logs

to execute it.

LRO80, 2021-03-23:
Hi, have you got a solution. 

I am getting the same fault using the same syntax like you. Instead of rsa I also tried a new created key with ed25519, nothing changed. Its quiet frustrating. 

Cheers 

Lars

ksaadatg, 2021-05-23:
Hi, my problem was the following: Downloading/cloning dgiese/dustcloud will create several sub folders such as dustcloud/devices/xiaomi-vacuum... 
There is another firmware builder which is in conflict with step no. 6 from dgiese/dustcloud wiki.
The command in step 6 must be shortend, so it refers to the correct firmware builder, the correct key etc.
In my case the command was --> "sudo ./dustcloud/imagebuilder.sh -f v11_003468.fullos.pkg -k id_ed25519.pub" ... depending which firmware and key you are using. Good luck!

closed 2021-12-12 by dgiese