dgiese/dustcloud #300: build() takes at most 4 positional arguments (5 given)

aezzati, 2021-01-21:
I have the following code, which fives me an error. Is this related to the Jupyter notebook version that I'm using?

random_state = 20

X_train, X_test, y_train, y_test = train_test_split(
    Xt, y, test_size=0.25, random_state=random_state)

rsf = RandomSurvivalForest(n_estimators=1000,
                           min_samples_split=10,
                           min_samples_leaf=15,
                           max_features="sqrt",
                           n_jobs=-1,
                           random_state=random_state)
rsf.fit(X_train, y_train)

TypeError                                 Traceback (most recent call last)
<ipython-input-4-ca18dfaa0802> in <module>
      5                            n_jobs=-1,
      6                            random_state=random_state)
----> 7 rsf.fit(X_train, y_train)

/Applications/anaconda3/lib/python3.8/site-packages/sksurv/ensemble/forest.py in fit(self, X, y, sample_weight)
    291             # parallel_backend contexts set at a higher level,
    292             # since correctness does not rely on using threads.
--> 293             trees = Parallel(n_jobs=self.n_jobs, verbose=self.verbose,
    294                              **_joblib_parallel_args(prefer='threads'))(
    295                 delayed(_parallel_build_trees)(

/Applications/anaconda3/lib/python3.8/site-packages/joblib/parallel.py in __call__(self, iterable)
   1059 
   1060             with self._backend.retrieval_context():
-> 1061                 self.retrieve()
   1062             # Make sure that we get a last message telling us we are done
   1063             elapsed_time = time.time() - self._start_time

/Applications/anaconda3/lib/python3.8/site-packages/joblib/parallel.py in retrieve(self)
    938             try:
    939                 if getattr(self._backend, 'supports_timeout', False):
--> 940                     self._output.extend(job.get(timeout=self.timeout))
    941                 else:
    942                     self._output.extend(job.get())

/Applications/anaconda3/lib/python3.8/multiprocessing/pool.py in get(self, timeout)
    769             return self._value
    770         else:
--> 771             raise self._value
    772 
    773     def _set(self, i, obj):

/Applications/anaconda3/lib/python3.8/multiprocessing/pool.py in worker(inqueue, outqueue, initializer, initargs, maxtasks, wrap_exception)
    123         job, i, func, args, kwds = task
    124         try:
--> 125             result = (True, func(*args, **kwds))
    126         except Exception as e:
    127             if wrap_exception and func is not _helper_reraises_exception:

/Applications/anaconda3/lib/python3.8/site-packages/joblib/_parallel_backends.py in __call__(self, *args, **kwargs)
    593     def __call__(self, *args, **kwargs):
    594         try:
--> 595             return self.func(*args, **kwargs)
    596         except KeyboardInterrupt as e:
    597             # We capture the KeyboardInterrupt and reraise it as

/Applications/anaconda3/lib/python3.8/site-packages/joblib/parallel.py in __call__(self)
    260         # change the default number of processes to -1
    261         with parallel_backend(self._backend, n_jobs=self._n_jobs):
--> 262             return [func(*args, **kwargs)
    263                     for func, args, kwargs in self.items]
    264 

/Applications/anaconda3/lib/python3.8/site-packages/joblib/parallel.py in <listcomp>(.0)
    260         # change the default number of processes to -1
    261         with parallel_backend(self._backend, n_jobs=self._n_jobs):
--> 262             return [func(*args, **kwargs)
    263                     for func, args, kwargs in self.items]
    264 

/Applications/anaconda3/lib/python3.8/site-packages/sklearn/ensemble/_forest.py in _parallel_build_trees(tree, forest, X, y, sample_weight, tree_idx, n_trees, verbose, class_weight, n_samples_bootstrap)
    167                                                         indices=indices)
    168 
--> 169         tree.fit(X, y, sample_weight=curr_sample_weight, check_input=False)
    170     else:
    171         tree.fit(X, y, sample_weight=sample_weight, check_input=False)

/Applications/anaconda3/lib/python3.8/site-packages/sksurv/tree/tree.py in fit(self, X, y, sample_weight, check_input, X_idx_sorted)
    237                                            params["min_impurity_split"])
    238 
--> 239         builder.build(self.tree_, X, y_numeric, sample_weight, X_idx_sorted)
    240 
    241         return self

sklearn/tree/_tree.pyx in sklearn.tree._tree.DepthFirstTreeBuilder.build()

TypeError: build() takes at most 4 positional arguments (5 given)

felipe0216, 2021-03-17:
I have the same issue and I am using the code suggested by the documentation: 
https://scikit-survival.readthedocs.io/en/latest/user_guide/random-survival-forest.html

aezzati, 2021-03-17:
Actually, I figured out the issue: You need to use the exact same version of Python. I switched to version 3.7 and it solved the issue.

closed 2021-12-12 by dgiese