dgiese/dustcloud #305: Wiki: CCC-to-CE for STYJ02YM

k-0, 2021-03-09:
Hi,

I tried to do this:
https://github.com/dgiese/dustcloud/wiki/Vacuum-Robots-CCC-to-CE-conversion

But my STYJ02YM (v7 - chinese version) has no `default `partition:
```
root@TinaLinux:~# cd /mnt/ && ls -l
drwxrwxrwx    3 root     root          7168 Jan  1  1970 SNN
drwxr-xr-x    7 root     root          1024 Mar 10 06:01 UDISK

```
Is there a way to change the v7 to v8 (global version)?

k-0, 2021-03-10:
I found no roborock.conf on the file-system, but the `/etc/miio/device.conf` looks interesting:
(Anonymized did, key and mac)
```
did=375500000 
key=WjVp000000000000
vendor=viomi
mac=5C:6B:00:00:00:00 
model=viomi.vacuum.v7
```
What would happen if I e.g. change it to `model=viomi.vacuum.**v8**` ?

Subtixx, 2021-03-10:
The logserver is saved in /mnt/UDISK/config/logserver.conf

dgiese, 2021-03-10:
Hi,
the viomi robots (v7,v8, etc) are not the same as the Rockrobo robots. The software is completely different (openwrt vs. ubuntu). Not sure how the region mechanism exactly works, but I dont think that there is a functional difference between v7 and v8. But the firmware is different.... so if there are differences, they are likely in firmware... and v8 firmware is not compatible with the one from v7 (I think it has different partitions...). What are you trying to archive?

k-0, 2021-03-10:
My intention is that I can connect the robot to Amazon Alexa. This is possible with the global version, but not with the Chinese version.

dgiese, 2021-12-12:
For reference: The Viomi robots seem to be EOL, as they did not get any updates for a very long time. Over builder.dontvacuum.me I have a conversion firmware that converts a v7/v8 to a v6. However, I have no clue if that actually solves your problem with Alexa.

closed 2021-12-12 by dgiese