dgiese/dustcloud #318: Discussion: S6 MaxV UART pins

don-code, 2021-12-18:
I recently bought a Roborock S6 MaxV and am trying to figure out where (if) I can find UART on the board.

The board on the S6 MaxV is substantially different from the board on the S6, shown in #213. I found what could be same cluster of pins (except TPA8 is TPA4). Those pins are nearby the SoC, so that might make sense - but it has a rubber pad over it, so I can't tell which it is. It's not Wi-Fi, as that's on a daughterboard.
[image]

TPA4 (what would have been 8 / TX on the S6) and TPA15 both show 3.3v during boot.

TPA16 (maybe the same on the S6?) shows some mystery 500khz signal, which is the wrong frequency for 115200 baud (or really any common baud rate - I can't set my UART to that). I'd normally accept that it's just not the TX pin, but this is the only test point anywhere on the board that's not giving me straight DC.

[image]

The actual TPA8 on this board is ground - it's in the upper right of the photo.

Any pointers on either what I'm looking at with the mystery signal, or where I can find the UART pins, would be greatly appreciated.

ky-devblog, 2022-03-05:
I was finding what is debug port( tx rx gnd) of the s6 maxv.
thank you for your contribution. It was helpful.

mottscanley, 2025-08-21:
@don-code it's a long shot but did you ever find out more about UART pins? Even longer shot, but I'm not even trying to root, just trying to see if there's any output on serial in order to diagnose why power led bink rapidly and if there's any way to fix seemingly bootlooping board..

amerwail00-cloud, 2026-09-25:
@mottscanley 
im having the same white light blinking issue in saros 20 sonic, this same issue happened to 2 units of saros 20 and one z70 and one curve 2 pro. 
ive been trying to find a way to access the boot log to find what is causing the board to boot loop. 
if found anything helpful i would be grateful.