dgiese/dustcloud #323: Roborock S5 Max firmware ver 1566 11/2021 confirmed rootable

Lanchon, 2022-02-20:
hi,

just a quick note to let you know that the latest S5 Max stock firmware (ver 1566 11/2021) is still rootable by UART+bootloader interruption+init=/bin/sh.

i installed dustbuilder-built fw (same version) and tested root via ssh and adb and both work without issue, as does everything else.

thanks!!

broizter, 2022-05-05:
After reading this post I did some research and it seems like multiple people have upgraded to 1566 without issue, and I can confirm it worked for me too. Not sure how the verification process works but maybe it's time to remove "might impact rooting" from the dustbuilder website?

Lanchon, 2022-05-05:
"might impact rooting" probably means untested for now, which is better than not saying anything.
but @dgiese has obviously been busy with other stuff so far. life happens when you open source... :)

Handy-E, 2022-10-20:
Seems fw-version 4.1.2_1642 is out (since 08 2022). Is it rootable and will dustbuilder offer it?
Currently I am on 1566 and upgrading from rooted 1482 worked for me too.

dgiese, 2022-10-21:
the big question is if they added something that prevents us from rooting or does something weird. I might test it in the next few weeks.

closed 2024-02-08 by Lanchon