dgiese/dustcloud #325: Rooting Xioami MJSTG1

phodina, 2022-03-16:
Hi,

I've got question if somebody tried to root the Xioami Robot Vacuum Mop Essential?

As I do own one I disassembled the cleaner and looked inside. Here are my observations:

- There is custom processor AM308S which I do not recognize and even google has not returned anything useful. Without some datasheet is hard to guess the pinout of the chip (without reverseengineering the board). There is no STM32 for driving the peripherals. Just the main SoC and Winbond nonvolatile memory.

- When I disconnected the mainboard R1_MainBoard_V1.4 (2019-12-27) I can see a lot of test points on the back of it. Though I have no Idea which is the UART or even if it's enabled.

- There is also attached ESP-WROOM-32D for bluetooth and wifi.

- This model does not have LIDAR nor up facing camera and relies only on infrared sensors. This makes is really affordable.

- The `python-miio` is able to recognize this device as the class is already implemented 
https://github.com/rytilahti/python-miio/blob/master/miio/integrations/vacuum/mijia/g1vacuum.py

So what should be the next steps to root this device?

Also any idea where to get the OTA firmware so that it could be analyzed (SoC architecture, security features ...)

phodina, 2022-03-16:
Here are some images of the robot:

[image]

[image]

Main board (has lots of tests points):

[image]

The connectors in the corners without soldered components look like USB:

[image]

[image]
[image]
[image]

The ESP32 is attached to the mainboard through the keyboard board:

[image]

Some pics of the robot as a whole:

[image]

[image]
[image]
[image]

phodina, 2022-03-18:
@dgiese  any ideas?

xedsvg, 2022-03-24:
@phodina I think this uses the ESP32 as the CPU not only for bluetooth and wifi. Try to play with that pcb, maybe a serial output will give out some hints.

phodina, 2022-03-25:
@xedsvg you mean there's actually no beefy processor running Linux but just tiny microcontroller like ESP32 running some RTOS or baremetal handling all the main tasks while the chip  on the mainboard AM308S handles all the IO?

xedsvg, 2022-03-25:
Don't take my word for it but that's **my guess.**

Linux would bloat the ESP32. You may be able to reverse engineer something from it tho. 
If I were you, I would poke all the uart pins and see what's the deal between the AM308S and the ESP.
If you are lucky you might get some uart comunication and expand on that. 
If that's not the case, see if you can dump the ESP. If it's encrypted use a [power glitch](https://raelize.com/blog/espressif-systems-esp32-bypassing-sb-using-emfi/).
If you can dump the firmware, search for strings inside of it, you may get the mqtt credentials. I would set up a mqtt proxy ([IOXI](https://github.com/NVISOsecurity/IOXY)) and replace them with the local proxy and see what's sending back.

philhug, 2022-10-16:
The ESP32 is only doing WIFI/cloud communication. The robot also works with the ESP32 removed.

philhug, 2022-10-29:
@xedsvg was able to extract the flash from the ESP32. no protection used. dump is readable.
ESP32 communicates with xiaomi cloud over tls-based protocol authenticating with device-specific keys. (derived from hardware, mac address?, fuses..?)
still need to find out what is sent over the UART between ESP and AM308S

philhug, 2022-10-30:
I installed esphome on the ESP32 and am able to control it with it :-)

philhug, 2022-10-31:
On the keyboard PCB there's debug port J1 which is exposed on the top of the robot if a plastic cover is removed. I haven't looked at the pinout though.

phodina, 2022-11-11:
Nice @philhug, haven't had the time. I disassembled the device but didn't do into the pins. 

Could you post the link to the firmware dump from the ESP32? I can I also dump mine and we can compare the differences.

phodina, 2022-11-11:
I've been looking into routers/modems capable running OpenWrt. I can divert the attention to ESP32. And to make the robot be controlled by the commandes recived by the Wifi/Bluetooth. It's dirt cheap and would open new world of IoT devices.

daniel-dona, 2023-02-16:
Looking into this robot, could be interesting to replace the ESP32 with something like an Orange Pi Zero or similar board and install there Valetudo.

philhug, 2023-02-16:
@daniel-dona no need to replace it, just flash esphome on it.

philhug, 2023-02-16:
Btw: there is also a debug port on the top side. I didn't look at the pinout yet, but I'd assume you could even flash it without disassembling it, like you have to do no.

daniel-dona, 2023-02-16:
Valetudo does not run on ESP32, is a Node.js based tool/interface, too heavy for that SoC. https://github.com/Hypfer/Valetudo

philhug, 2023-02-16:
I know, that's why I suggest to use esphome for the esp32.

daniel-dona, 2023-02-17:
Sure, but ESPhome is more like an interfacing software, Valetudo is an integrated dashboard... they can even integrate if Valetudo is installed outside the robot.

daniel-dona, 2023-02-17:
Some more information about this robot. The main chip is this one http://en.amicro.com.cn/?product/chip/master/1675.html

Looks like a purpose build SoC for vacuum cleaning robots with a custom SDK/RTOS (http://en.amicro.com.cn/?platform/open/), no Linux or something similar. Only 64 KB or SRAM anyway...

The debug port on the top looks like it's only an UART connection for debuging this Amicro AM380S chip. Nothing interesting but I logged it here: https://github.com/daniel-dona/xiaomi_MJSTG1_hacking/blob/main/main_bootlog.txt

philhug, 2023-02-17:
Thanks for the log+finding the SoC.
I'm  using the ESP32  mostly as an UART over Wifi bridge, so it would also be possible to run Valetudo externally.

The main thing I am missing at the moment are the commands for the map. Start/stop/dock are easy.

philhug, 2023-02-17:
@daniel-dona I just looked at your dumps (ipc dump is what I am interested in) can you do a dump that contains some map updates during cleaning?
I replaced the firmware too early :-)))

daniel-dona, 2023-02-18:
Some more information about this robot. The main chip is this one http://en.amicro.com.cn/?product/chip/master/1675.html

Looks like a purpose build SoC for vacuum cleaning robots with a custom SDK/RTOS (http://en.amicro.com.cn/?platform/open/), no Linux or something similar. Only 64 KB or SRAM anyway...

The debug port on the top looks like it's only an UART connection for debuging this Amicro AM380S chip. Nothing interesting but I logged it here: https://github.com/daniel-dona/xiaomi_MJSTG1_hacking/blob/main/main_bootlog.txt

> @daniel-dona I just looked at your dumps (ipc dump is what I am interested in) can you do a dump that contains some map updates during cleaning? I replaced the firmware too early :-)))

I don't think so, the robot was just in my desk during the capture, but I will try to capture a whole cleaning session. Anyway I'm not sure the map is used for anything more than eye-candy in the Xiaomi App... no LiDAR or V/SLAM in this robot, so is more like the log of all the bumps with things around the house than something that is used during the cleaning session.

daniel-dona, 2023-02-18:
> > I don't think so, the robot was just in my desk during the capture, but I will try to capture a whole cleaning session. Anyway I'm not sure the map is used for anything more than eye-candy in the Xiaomi App... no LiDAR or V/SLAM in this robot, so is more like the log of all the bumps with things around the house than something that is used during the cleaning session.
> 
> Yeah, but it still builds some kind of map that resembles the room. Also the reads/writes are somehow buffered in your log, so the commands are not in the correct order. it should look like this:
> 
> ```
> > get_down
> < down none
> > get_down
> < down ...
> ```

That is because I captured only one line at a time, TX from the SoC or TX from the ESP32. I only have a cheap UART to USB adapter in hand currently :/

Petro0872, 2024-03-10:
HI! 
Do you have a  firmware for ESP32 or dump? I need a dump to write flash.

philhug, 2024-03-10:
@Petro0872 If you can write the flash you should also be able to dump it yourself.
What's the issue?

Petro0872, 2024-03-10:
The original flash is break. I replace  it with a new spi flash and now i need a dump to write a new flash.

ryannza, 2025-10-17:
Hi all. Great work. 
I've also stumbled down this path, although to repurpose as a DIY robot with PS4 controller option.
I noticed tx2 and rx2 from esp-wroom to the mainboard. 
Unfortunately this was too late because I already reset the original wroom. 
I struggled a bit with programming. Also noticed the wroom is single core locked by efuse.
If anybody could share the original firmware, I would really appreciate that. 
Otherwise I have an esp32-c3 super mini that I will try and use for replacement for now, but I think I'm going to have hurdles with communication to the mainboard. 
If that fails, my fallback plan is to replace the mainboard with a dual core esp32, especially since the apparent brushed wheel motors seem to have drivers already.

draconx2018, 2026-08-25:
Hello, I have a backup of the original firmware, but I also have a problem, whatever I write on the esp doesn't work, it goes into a restart loop. Do you have any idea what it could be? I tried several processor architectures:
E (444) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
E (444) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig

draconx2018, 2026-08-28:
Yes, is correct, iI resolve it with this parameter
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: "y"
      CONFIG_ESP32_BROWNOUT_DET: "n"
    advanced:
      ignore_efuse_mac_crc: true
      ignore_efuse_custom_mac: true

draconx2018, 2026-09-06:
I've almost reached the end, but I would need your help, at what baudrate does the esp32 communicate with the R1_KEYPCB_V1.2 board and with the AM308S? I only receive the crap through that serial rx2/tx2

draconx2018, 2026-09-06:
@philhug Can you please help me with this information?

ggirodda, 2026-09-16:
Hi everyone,
I was looking to intercept the communication between the esp32 and the AM380S chip. I'm currently doing this on a Rowenta X-Plorer Serie 45 robot vacuum, it seems the hardware is similar to the xiaomi one (esp32 + am380s).
Here's the sniffer code and the link to the wiring diagram:
https://github.com/ggirodda/rowenta-xplorer-s45-sniffer
The sniffer code has mostly been generated by AI, but it works.
What I'm missing on the Rowenta app, is the possibility to manually give directions to the robot, which that means I can't intercept these commands for me to be able to reproduce them.
Can anyone help me and sniff the traffic between the xiaomi esp32 and the am380s while giving manual direction? This would be really helpful.
Thank you :)
EDIT. I forget to add that I also successfully can communicate with the AM380S chip by using the sniffed commands, start, stop, back to the dock, buzz command are successfully working