dgiese/dustcloud #331: Support for Dreame W10 rooting

DevelopingEntitiesWithFuntations, 2023-01-21:
When I try to root the Dreame W10 according to the instructions when I click on the reset button for any amount an error pops up showing that `/etc/os-release` is missing and no login prompt appears...

My boot log looks like the following:

```
[168]HELLO! BOOT0 is starting!
[172]boot0 commit : efb321e8df5f8e8bf2fc37c14f800ab5f54b12e5
 
[217]key not pressed anymore
MESSAGE: [0x0] TEE-CORE: arisc version: [v0.3.72]

[1.880]
Starting kernel ...

e2fsck 1.42.12 (29-Aug-2014)
ext2fs_check_if_mount: Can't check if filesystem is mounted due to missing mtab file while determining whether /dev/by-name/misc is mounted.
/dev/by-name/misc: clean, 16/256 files, 1112/2048 blocks
Failed to connect to ubus
oggdec from vorbis-tools 1.4.0
Decoding "/audio/EN/0.ogg" to "/tmp/0.wav"
        [100.0%]
Playing WAVE '/tmp/0.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
crond[1092]: crond (busybox 1.31.0) started, log level 8

p2027
killall: rsyslogd: no process killed
chpasswd: password for 'root' changed
channel:6 is ok
Configuration file: /tmp/hostapd.conf
rfkill: Cannot open RFKILL control device
Using interface wlan0 with hwaddr MACADDRESSHERE and ssid "dreame-vacuum-p2027_miap01523"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
p2027
{"ret":"ok"}
```
Please note that the SSID ID at the end and the MAC have been changed or censored.

cornfeedhobo, 2023-02-11:
I believe this means that your firmware is too new to attempt this method. You will have to try another method.

explo910, 2023-02-26:
I have the same problem. Are there any other public methods like downgrading the firmware for the W10?
Has anyone found anything on this subject with more luck than me ?